espressif/esp-idf #878: sd card example errors

Gfast2, 2017-08-04:
Hi espressif
Right now I'm working on how to interfacing with sd card. I found the example "sd_card" in folder ```example```/```storage```/```sd_card```.
Following all the instrauctions in its README, I wiring all things up including the pull-up resistors. I've only compile & upload the example code, and don't change it at all firstly. But I got the error as following:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57
```
It looks like I interrupt the communication between builtin SPI and CPUs. If I remove the 3.3V pin for all the pullup resistors. Then I got the following error:

```
I (1198) example: Initializing SD card
D (1198) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83
D (1208) sdmmc_periph: slot=1 div=50 freq=400kHz
D (1208) sdmmc_periph: slot=1 width=1
D (1218) sdmmc_cmd: sdmmc_card_init
D (1228) sdmmc_req: process_command_response: error 0x109  (status=00000042)
D (1228) sdmmc_req: process_command_response: error 0x109  (status=00000004)
E (1228) sdmmc_cmd: sdmmc_card_init: send_if_cond (1) returned 0x109
D (1238) vfs_fat_sdmmc: sdmmc_card_init failed 0x(109)
E (1248) example: Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.
```
In order to make sure I do wiring everything up correctly, I built the same circuit on another breadboard, but I get the same effect. 
The board I'm using is NodeMCU 32, It is a wrapper for a standard ESP-WROOM-32 module. I solder wires direct on a microSD card adapter. I've changed different µSD card to prevent potential problem. But it gives me the same error.

Gfast2, 2017-08-04:
I don't know if this will bring something:
When ESP32 is experiencing the first error, and I remove the 3.3V pin for all pull-up resistors. I got the second error immediately.

Gfast2, 2017-08-04:
BTW, I wanna know how the pin1 on a SD card works. according to [this](http://elasticsheep.com/2010/01/reading-an-sd-card-with-an-atmega168/) post. it have two functionality: ```CD/DAT3```. When I let all SD pins pullup, I got the first error above, if in this situation I only remove the pullup resistor on this pin, I got the second error Iike I reomve the voltage of all pullup resistors.

negativekelvin, 2017-08-04:
A lot of people having trouble to understand strapping pins and using for other functions.  Strapping pins **must** have certain state at boot and then can be pulled up/down **after** boot using some type of time delay circuit.  Some can be bypassed with efuse setting.

Gfast2, 2017-08-04:
I found this [Answer](http://www.edaboard.com/thread209780.html) about ```strapping pins```
> To assign a default state of a pin on power on or reset time.
> When do we use : If a logic value '0' or '1' is an active value of this pin, and you don't want to be activated it on power-on or reset,
> you may strap it to a value that you need...
Finally I figured out what is the problem: For ```GPIO2``` and ```GPIO12``` I used 23KΩ resistor. It seems too big to actually does its job. But I'm not sure about this, in order to make sure my home made µSD card adapter works correctly, I've change my ESP32 board from NodeMCU32 (ESP_WROOM_32 Wrapper) to SparkFun ESP32 Thing (DEV-13907). I change these pullup resistor with 330Ω resistors (I don't have 10KΩ ones right now) and everything works just fine. ;D

closed 2017-08-04 by Gfast2