espressif/esp-idf #879: Bluetooth classic pairing

OtherCrashOverride, 2017-08-04:
After looking through the documentation and Bluetooth examples, it is not clear how to securely pair the ESP32 with another device over Bluetooth.  Is classic pairing supported? If so, is there documentation or examples?

For clarification, the expectation is that when an ESP32 is connected to over Bluetooth, a pairing code should be entered.  This is a basic security requirement to secure devices from unauthorized access.

FHFS, 2017-08-07:
Have a look at [secure gatt](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c)

The pairing code can be shown if the device io capability is set to display (ESP_IO_CAP_OUT)
If ESP_IO_CAP_NONE is set the justworks association model will be used.

OtherCrashOverride, 2017-08-14:
I tested with the ESP_IO_CAP_OUT setting.  It appears to have no effect on Bluetooth Classic A2DP.  Any device can still connect to the ESP32 without any PIN number.

FHFS, 2017-08-14:
I used GATTC to read write characteristics and i had to set:
```
esp_ble_gattc_read_char(gattc_if, conn_id,
									&provision_id,
									&next,
									ESP_GATT_AUTH_REQ_SIGNED_NO_MITM);
```
this way it encrypts the data read/written. I dont know about a2dp.

wangwangwang1998, 2017-10-09:
The `a2dp_sink` demo doesn't have encrypt API now.  
If you want to connect to the ESP32 with PIN number,you can use the `gatt_security_client` and `gatt_security_server` demo.

OtherCrashOverride, 2017-10-17:
Was there a resolution to this issue?

blueMoodBHD, 2018-03-21:
Now, there is no API to set the device io capability, we just set IO capacity as noinputnooutput, so you can connect to other devices without pin code. And, we plan to add the APIs in IDF release v3.1. Thanks.

blueMoods, 2018-07-16:
Hi all,
This feature has merged into Master, now. Refer to commit: 8571170

blueMoodBHD, 2018-07-16:
Hi all,
This feature has merged into Master, now. Refer to commit: 8571170