espressif/esp-idf #880: ASSERT_PARAM(131072 0), in rwble.c at line 222

SiarheiVolkau, 2017-08-05:
Hello all,

We are working on BLE enabled device, and rarely we have getting this error:
`ASSERT_PARAM(131072 0), in rwble.c at line 222`
We can't reproduce this error.
Our setup:
ESP-WROOM-32 `Chip is ESP32D0WDQ6 (revision 0)`
ESP-IDF: revision `9b955f4`

We are want to know what happened here and how tho fix that, since we not have codes of rwble.c we can't understand root cause of this issue.

SiarheiVolkau, 2017-08-05:
The main issue here -- board not rebooted even if watchdog(s) are enabled.
If we can't fix issue itself we want to make board rebooting when it happened again.

jkhax0r, 2017-08-05:
I've seen the same issue once or twice.

Right now I have it very repeatably doing the same but it says:

[code] ASSERT_PARAM(64 0), in rwble.c at line 222 [/code]

This is going to kill my project so any help would be appreciated

Happens if I leave things running for 4 hours or so.  

My BLE use case is one GATTS profile running (unconnected) and 1 connected GATTC device that is not sending any data besides maintaining a connection with interval=6 (7.5ms) sleep latency=499 and timeout=3000 (30s).  It's based on a CC2540 chipset device, which I have seen lots of issues with.

SiarheiVolkau, 2017-08-06:
We also use CC2540 as Peripherals, when ESP32 used as Central.

limooi, 2017-08-14:
Yes experiencing same issue on the same ESP-WROOM32 as SiarheiVolkau

TianHao-Espressif, 2017-09-26:
@jkhax0r , We are doing fix "ASSERT_PARAM(64 0), in rwble.c at line 222", and other error may related to this error.

mywang-espressif, 2017-10-16:
Hi @jkhax0r , there is a bug fix for the issue "ASSERT_PARAM(64 0), in rwble.c at line 222" in some cases. The commit ID is 3106ca1c1e7735e88deb77c6fd4ce5721a28b2b5, please have a try and see whether there is improvement.

ArTyGill, 2018-08-09:
Yes experiencing same issue on the same ESP-WROOM32 as limooi

My error is:

ASSERT_PARAM(32 0), in rwble.c at line 222??????

chegewara, 2018-11-23:
@mywang-espressif @Weijian-Espressif 
There is few issues still pending about `rwble.c`, but i post here because in this issue we have info that its been fixed already. Here is my log from wrover board:
```
ASSERT_PARAM(512 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40087058  PS      : 0x00060034  A0      : 0x8008a602  A1      : 0x3ffbe2a0
0x40087058: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffbdcd0  A7      : 0x60008050  A8      : 0x80087055  A9      : 0x3ffbe280
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40086f8d  LEND    : 0x40086f94  LCOUNT  : 0x00000000
0x40086f8d: r_assert_param at ??:?

0x40086f94: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x4018e0bc  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087058
0x4018e0bc: ip4addr_aton at D:/msys32/home/imper/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_addr.c:247 (discriminator 3)

0x40087058: r_assert_param at ??:?

Backtrace: 0x40087058:0x3ffbe2a0 0x4008a5ff:0x3ffbe2c0 0x4008b1cb:0x3ffbe2e0 0x40084dbd:0x3ffbe300 0x4018e0b9:0x00000000
0x40087058: r_assert_param at ??:?

0x4008a5ff: r_rwble_isr at ??:?

0x4008b1cb: r_rwbtdm_isr_wrapper at intc.c:?

0x40084dbd: _xt_lowint1 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

0x4018e0b9: ip4addr_aton at D:/msys32/home/imper/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_addr.c:247 (discriminator 3)

Core 1 register dump:
PC      : 0x401d256e  PS      : 0x00060a34  A0      : 0x800d3e65  A1      : 0x3ffbbac0
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x7fffffff  A5      : 0x00000001
A6      : 0x00060420  A7      : 0x00000000  A8      : 0x800d2ed6  A9      : 0x3ffbba90
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x800942d6  A13     : 0x3ffbb9d0
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x401d256e:0x3ffbbac0 0x400d3e62:0x3ffbbae0 0x400936b9:0x3ffbbb00 0x400940dd:0x3ffbbb20
0x400d3e62: psram_set_basic_write_mode at D:/msys32/home/imper/esp/esp-idf/components/esp32/spiram_psram.c:149 (discriminator 10)

0x400936b9: rtc_clk_bbpll_configure at D:/msys32/home/imper/esp/esp-idf/components/soc/esp32/rtc_clk.c:474

0x400940dd: rtc_init at D:/msys32/home/imper/esp/esp-idf/components/soc/esp32/rtc_init.c:68

CPU halted.
```
What does my app:
- is working as server that is serving bin file to ota over ble,
- few hour before crash its been just sitting and advertising own presence, before crash i updated other esp32 from that esp32 wrover with success
- in the same time esp http client is running, but it is used only to download bin file from server when esp32 starts, nevertheless both wifi and ble are running all the time

chegewara, 2018-11-24:
I have one more app crash probably not related, but it happened when i restarted esp32 after above crash. One more time logs from above crash, followed by crash after reset few hours after crash, and again crash after second reset (all reset procedures from make monitor):
```
ASSERT_PARAM(512 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40087058  PS      : 0x00060034  A0      : 0x8008a602  A1      : 0x3ffbe
2a0
0x40087058: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008
054
A6      : 0x3ffbdcd0  A7      : 0x60008050  A8      : 0x80087055  A9      : 0x3ffbe
280
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xfffff
fff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000
005
EXCVADDR: 0x00000000  LBEG    : 0x40086f8d  LEND    : 0x40086f94  LCOUNT  : 0x00000
000
0x40086f8d: r_assert_param at ??:?

0x40086f94: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087
058
0x40087058: r_assert_param at ??:?

Backtrace: 0x40087058:0x3ffbe2a0 0x4008a5ff:0x3ffbe2c0 0x4008b1cb:0x3ffbe2e0 0x4008
4dbd:0x3ffbe300 0x4000bfed:0x00000000
0x40087058: r_assert_param at ??:?

0x4008a5ff: r_rwble_isr at ??:?

0x4008b1cb: r_rwbtdm_isr_wrapper at intc.c:?

0x40084dbd: _xt_lowint1 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xte
nsa_vectors.S:1105

Core 1 register dump:
PC      : 0x401d256e  PS      : 0x00060334  A0      : 0x800d3e65  A1      : 0x3ffbb
ac0
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x800942d6  A5      : 0x3ffbb
9f0
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d2ed6  A9      : 0x3ffbb
a90
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x800942d6  A13     : 0x3ffbb
9d0
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000
005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000
000

Backtrace: 0x401d256e:0x3ffbbac0 0x400d3e62:0x3ffbbae0 0x400936b9:0x3ffbbb00 0x4009
40dd:0x3ffbbb20
0x400d3e62: psram_set_basic_write_mode at D:/msys32/home/imper/esp/esp-idf/componen
ts/esp32/spiram_psram.c:149 (discriminator 11)

0x400936b9: rtc_clk_bbpll_configure at D:/msys32/home/imper/esp/esp-idf/components/
soc/esp32/rtc_clk.c:474

0x400940dd: rtc_init at D:/msys32/home/imper/esp/esp-idf/components/soc/esp32/rtc_i
nit.c:68

CPU halted.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
[... 371 lines cut ...]
0x40094408: rtc_clk_wait_for_slow_cycle at D:/msys32/home/imper/esp/esp-idf/compone
nts/soc/esp32/rtc_time.c:153

0x40092e7b: disableAllWdts at D:/msys32/home/imper/esp/esp-idf/components/esp32/pan
ic.c:649
 (inlined by) commonErrorHandler at D:/msys32/home/imper/esp/esp-idf/components/esp
32/panic.c:587

0x4008b7ca: r_rwip_sleep at ??:?

0x4008d1e5: coex_arbit_insert at ??:?

0x4008c8fb: coex_timer_ts_end_handler at ??:?

0x4008fbb3: xTaskResumeAll at D:/msys32/home/imper/esp/esp-idf/components/freertos/
tasks.c:4458

0x40084dbd: dram_alloc_to_iram_addr at D:/msys32/home/imper/esp/esp-idf/components/
heap/heap_caps.c:345

0x400862d6: get_mmu_region at D:/msys32/home/imper/esp/esp-idf/components/spi_flash
/flash_mmap.c:319

CPU halted.
```

JosuGZ, 2019-02-14:
Why is this closed?

Alvin1Zhang, 2019-02-14:
@JosuGZ Do you have similar issue mentioned in this page? How is the status of your issue? Thanks

JosuGZ, 2019-02-14:
> @JosuGZ Do you have similar issue mentioned in this page? How is the status of your issue? Thanks

I'm getting an `ASSERT_PARAM(131072 0), in rwble.c at line 230` after a while, scanning BLE beacons.

I'm currently on 'v3.3-beta1-223-ga62cbfec9'

JosuGZ, 2019-02-15:
This was the log:

```
ASSERT_PARAM(131072 0), in rwble.c at line 230
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008576d  PS      : 0x00060634  A0      : 0x800890c4  A1      : 0x3ffbe310
0x4008576d: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffbdccc  A7      : 0x60008050  A8      : 0x8008576d  A9      : 0x3ffbe2f0
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400856a5  LEND    : 0x400856ac  LCOUNT  : 0x00000000
0x400856a5: r_assert_param at ??:?

0x400856ac: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40087b71  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008576d
0x40087b71: r_lld_evt_end_isr at ??:?

0x4008576d: r_assert_param at ??:?

Backtrace: 0x4008576d:0x3ffbe310 0x400890c1:0x3ffbe330 0x40089ca7:0x3ffbe350 0x40083bb2:0x3ffbe370 0x4000bfed:0x00000000
0x4008576d: r_assert_param at ??:?

0x400890c1: r_rwble_isr at ??:?

0x40089ca7: r_rwbtdm_isr_wrapper at intc.c:?

0x40083bb2: _xt_lowint1 at C:/Users/Josu/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

CPU halted.
```

I believe it works fine on the release/v3.2 though, I think I was on a recent commit on Master, one of these:

- bf1955aac819f501431f6d2230885ce8a4d4e6b8
- a62cbfec9ab681bddf8f8e45d9949b1b1f67a2ec

I'm seeing that if something takes too long on the ble callback, some timeout watchdog will trigger.

Alvin1Zhang, 2021-09-16:
Thanks for reporting, fix is available at a4c513b1, feel free to reopen.

closed 2021-09-16 by Alvin1Zhang

SiarheiVolkau, 2021-09-16:
@Alvin1Zhang, I didn't found the `a4c513b1` commit you mention anywhere, could you please post the link to it here?

Wth-Esp, 2021-09-16:
@SiarheiVolkau  IDF commit [e16ec9a574] fixed.