espressif/esp-idf #883: Random crash when using bluetooth and wifi

SquallHalle, 2017-08-06:
I am try to use https://github.com/MrBuddyCasino/ESP32_MP3_Decoder with both, wifi connection and as BT-AudioSpeaker. i modified the code so wifi and the BT-Audio works. After a random wihle i got this:

```
--- idf_monitor on COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (15) boot: ESP-IDF v3.0-dev-214-ge1653364 2nd stage bootloader
I (15) boot: compile time 16:25:14
I (16) boot: Enabling RNG early entropy source...
I (30) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (68) boot: Partition Table:
I (79) boot: ## Label            Usage          Type ST Offset   Length
I (101) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (124) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (148) boot:  2 factory          factory app      00 00 00010000 00200000
I (171) boot: End of partition table
I (184) boot: Disabling RNG early entropy source...
I (201) boot: Loading app partition at offset 00010000
I (219) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x437e4 (276452) map
I (541) esp_image: segment 1: paddr=0x0005380c vaddr=0x3ffc0000 size=0x03138 ( 12600) load
I (557) esp_image: segment 2: paddr=0x0005694c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (563) esp_image: segment 3: paddr=0x00056d54 vaddr=0x40080400 size=0x092bc ( 37564) load
I (637) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xbc4a8 (771240) map
0x400d0018: _stext at ??:?

I (1460) esp_image: segment 5: paddr=0x0011c4c8 vaddr=0x400896bc size=0x0c5f4 ( 50676) load
0x400896bc: panicHandler at C:/msys32/home/Username/esp/esp-idf/components/esp32/panic.c:546

I (1525) esp_image: segment 6: paddr=0x00128ac4 vaddr=0x400c0000 size=0x00000 (     0) load
I (1565) cpu_start: Pro cpu up.
I (1566) cpu_start: Starting app cpu, entry point is 0x40080f6c
0x40080f6c: call_start_cpu1 at C:/msys32/home/Username/esp/esp-idf/components/esp32/cpu_start.c:192

I (0) cpu_start: App cpu up.
I (1576) heap_init: Initializing. RAM available for dynamic allocation:
I (1596) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1615) heap_init: At 3FFD4200 len 0000BE00 (47 KiB): DRAM
I (1635) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1654) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1674) heap_init: At 40095CB0 len 0000A350 (40 KiB): IRAM
I (1694) cpu_start: Pro cpu start user code
I (1752) cpu_start: Starting scheduler on PRO CPU.
I (189) cpu_start: Starting scheduler on APP CPU.
I (189) main: starting app_main()
I (189) main: RAM left: 161356
I (269) main: hardware initialized
I (269) main: starting network
I (269) wifi: wifi firmware version: 0e33bec
I (269) wifi: config NVS flash: enabled
I (279) wifi: config nano formating: disabled
I (279) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (289) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (319) wifi: Init dynamic tx buffer num: 32
I (319) wifi: Init data frame dynamic rx buffer num: 32
I (319) wifi: Init management frame dynamic rx buffer num: 32
I (319) wifi: wifi driver task: 3ffd50c8, prio:23, stack:4096
I (329) wifi: Init static rx buffer num: 10
I (329) wifi: Init dynamic rx buffer num: 32
I (339) wifi: Init rx ampdu len mblock:7
I (339) wifi: Init lldesc rx ampdu entry mblock:4
I (339) wifi: wifi power manager task: 0x3ffe14a8 prio: 21 stack: 2560
I (359) wifi: wifi timer task: 3ffe24b0, prio:22, stack:3584
I (379) phy: phy_version: 357.0, a6bf95b, Jul 25 2017, 12:28:06, 0, 0
I (379) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (379) wifi: mode : sta (24:0a:c4:82:81:80)
I (389) wifi: received wifi event 2
I (389) wifi: connected successfully
I (509) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (1499) wifi: state: init -> auth (b0)
I (1499) wifi: state: auth -> assoc (0)
I (1499) wifi: state: assoc -> run (10)
I (1519) wifi: connected with Hai2, channel 11
I (1529) wifi: received wifi event 4
I (2259) event: ip: 192.168.1.223, mask: 255.255.255.0, gw: 192.168.1.1
I (2259) wifi: received wifi event 7
I (2259) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (2529) BT: config_parse returned with err code: 2

I (2569) renderer: init I2S mode 0, port 0, 16 bit, 44100 Hz
I (2569) I2S: queue free spaces: 1
I (2569) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32
I (2579) I2S: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, d
ivb: 11
I (2589) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (2589) main: RAM left 78464
I (2609) main: app_main stack: 2092

I (11509) wifi: pm start, type:0

E (13649) BT: btm_sec_conn_req

E (13659) BT: btm_sec_connected

E (15609) BT: btm_sec_conn_req

E (15619) BT: btm_sec_connected

I (16539) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (16539) BT_AV: a2dp conn_state_cb, state 1
I (16559) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x2
I (16559) BT_AV: a2dp audio_cfg_cb , codec type 0
I (16559) BT_AV: audio player configured
E (16639) BT: bta_dm_rm_cback:0, status:0
I (16639) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (16639) BT_AV: a2dp conn_state_cb, state 2
E (16689) BT: process_service_search_attr_rsp

E (16709) BT: bta_av_rc_opened allow incoming AVRCP connections:1
E (16709) BT: bta_av_rc_opened rcb[0] shdl:1 lidx:1/0
I (16719) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (16719) BT_AV: avrc conn_state evt: state 1, feature 0x24b, [10:d5:42:68:e9:66]
E (23519) BT: bta_av_proc_stream_evt: p_data is null
E (23519) BT: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x8202

E (23529) BT: bta_dm_rm_cback:0, status:0
E (23529) BT: bta_dm_rm_cback:0, status:7
E (23539) BT: BTA_AV_START_EVT status 0, suspending 0, init 0

I (23539) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x1
I (23549) BT_AV: a2dp audio_state_cb state 2
E (25559) BT_AV: audio data pkt cnt 100
I (26099) wifi: bcn_timout,ap_probe_send_start
E (27589) BT_AV: audio data pkt cnt 200
I (28609) wifi: ap_probe_send over, resett wifi status to disassoc
I (28609) wifi: state: run -> init (1)
I (28609) wifi: pm stop, total sleep time: 0/17102343

I (28609) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (28629) wifi: received wifi event 5
I (28789) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (28789) wifi: state: init -> auth (b0)
I (28809) wifi: state: auth -> assoc (0)
E (29589) BT_AV: audio data pkt cnt 300
I (29809) wifi: state: assoc -> init (4)
I (29809) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (29829) wifi: received wifi event 5
I (31569) wifi: received wifi event 5
E (31589) BT_AV: audio data pkt cnt 400
I (31689) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (31689) wifi: state: init -> auth (b0)
I (31699) wifi: state: auth -> assoc (0)
I (32699) wifi: state: assoc -> init (4)
I (32699) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (32709) wifi: received wifi event 5
E (33589) BT_AV: audio data pkt cnt 500
I (34429) wifi: received wifi event 5
E (35589) BT_AV: audio data pkt cnt 600
I (36189) wifi: received wifi event 5
E (37589) BT_AV: audio data pkt cnt 700
I (37949) wifi: received wifi event 5
E (39589) BT_AV: audio data pkt cnt 800
I (39689) wifi: received wifi event 5
I (41449) wifi: received wifi event 5
E (41589) BT_AV: audio data pkt cnt 900
I (43149) wifi: received wifi event 5
E (43589) BT_AV: audio data pkt cnt 1000
I (44889) wifi: received wifi event 5
E (45589) BT_AV: audio data pkt cnt 1100
I (46609) wifi: received wifi event 5
E (47609) BT_AV: audio data pkt cnt 1200
I (48329) wifi: received wifi event 5
I (48449) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (48459) wifi: state: init -> auth (b0)
I (49459) wifi: state: auth -> init (2)
I (49459) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (49479) wifi: received wifi event 5
E (49589) BT_AV: audio data pkt cnt 1300
I (51199) wifi: received wifi event 5
E (51589) BT_AV: audio data pkt cnt 1400
I (52929) wifi: received wifi event 5
ASSERT_PARAM(8192 0), in rwbt.c at line 360W (15) boot: PRO CPU has been reset by WDT.
W (15) boot: WDT reset info: PRO CPU PC=0x4008114b
0x4008114b: xt_highint4 at ??:?

W (15) boot: WDT reset info: APP CPU PC=0x40086e87
0x40086e87: uxPortCompareSet at C:/msys32/home/Username/esp/esp-idf/components/freertos/port.c:438
 (inlined by) vPortCPUAcquireMutex at C:/msys32/home/Username/esp/esp-idf/components/freertos/port.c:333

I (29) boot: ESP-IDF v3.0-dev-214-ge1653364 2nd stage bootloader
I (49) boot: compile time 16:25:14
I (62) boot: Enabling RNG early entropy source...
I (79) boot: SPI Speed      : 40MHz
I (91) boot: SPI Mode       : DIO
```

negativekelvin, 2017-08-06:
Related: https://github.com/espressif/esp-idf/issues/880

SquallHalle, 2017-08-07:
Sorry i dont the the connection. Different error, different behavior.

negativekelvin, 2017-08-08:
`ASSERT_PARAM(8192 0), in rwbt.c `
vs
`ASSERT_PARAM(131072 0), in rwble.c` 

Not the same error, but the fact they have been reported within a day of each other suggests they are related to some recent change in Bluetooth lib, possibly shared code

SquallHalle, 2017-08-08:
yes, on this way, i am with you.

petolat, 2017-09-29:
Hello,

Is there any update ?
I'm stuck with this bug because when it happens, the plateform doesn't reboot.

"ASSERT_PARAM(8192 0), in rwbt.c at line 360ets Jun  8 2016 00:22:57"

I'm on the latest esp-idf on master branch. I'm using the BT classic and wifi in promiscuous mode.

Thank you for your support.

TianHao-Espressif, 2017-10-19:
@negativekelvin , yes , may related to #880 , please try the commit ID after 3106ca1

sdaxini, 2017-11-01:
What about https://github.com/espressif/arduino-esp32 has it been merged or an eta?

closed 2017-11-21 by FayeY

SquallHalle, 2017-11-21:
So it is done ? or what ?

FayeY, 2017-11-22:
Hi @SquallHalle , have you tried the latest IDF? Is your problem still unsolved with the latest IDF?