espressif/esp-idf #884: Disconnect the wifi connection and reconnect to the wifi. After many such operations. "sento“ always fails. The errno is -1.

jixunx, 2017-08-07:
IDF is v2.0.Module is esp_wroom_32.The receiving terminal is smartphone. Smartphone as AP,esp32 as station. There is no other devices.Only one esp32 connects smartphone.The attachment is sample code,the sdkconfig file of sample code,and the log file.
Disconnect the wifi connection and reconnect to the wifi. After many such operations. "sendto“ always fails. The errno is -1 based on the return value is sendto is -1.
It is hard to find out what "errno is -1" indicates.So what's your opinion?Is my bug from my app level？
Best wishes！
[report.zip](https://github.com/espressif/esp-idf/files/1204046/report.zip)

liuzfesp, 2017-08-15:
Hi @jixunx could you let me know the IDF commit ID please? You can paste the startup log here.

jixunx, 2017-08-15:
The idfv2.0 and idfv2.1 are download from https://github.com/espressif/esp-idf/releases/.You could download  from the webpage.I don't know how to find commit ID.

liuzfesp, 2017-08-15:
Hi @jixunx you can paste the startup log here, the commit ID can be found in startup log, thanks!

jixunx, 2017-08-15:
The log is in the report.zip.But I can not find commit ID.The idfv2.0 and idfv2.1 version I download are released version.You could test with released version.You could found from  https://github.com/espressif/esp-idf/releases/.

zhangyanjiaoesp, 2017-08-25:
when “errno == -1”, it means the socket error is “ERR_IF”,

When “sendto()” function is called，the function reference relationship is as follws:
	(application layer)sendto()----→
	(lwip layer)lwip_sendto()----→netconn_send()-----udp_sendto()---->low_level_output()----→
	(mac layer)esp_wifi_internal_tx()(transmit the buffer via wifi driver)
	when user disconnect the wifi connection, the physical connection will disconnect first, at the same time, the application may execute to lwip_sento() or esp_wifi_internal_tx(), and then failed, the former return ERR_RTE, the latter return ERR_IF.
	These return errors is a formal phenomenon.
Note: 

1. When “sendto()” function is called to transmit UDP packet, it’s better to use “getsockopt()” function to get the value of “errno”, instead of printing it directly, because “errno” is a global variable, it may be impacted by other functions.
2.It’s better to use “ESP_LOGI” than use “printf” to print information on the screen.

I have modified your example code to make it better, it is in the attachment.
[report.tar.gz](https://github.com/espressif/esp-idf/files/1251149/report.tar.gz)

FayeY, 2017-10-10:
Hi

closed 2017-10-10 by FayeY

jixunx, 2017-11-02:
Thank you for your reply.But once we met "errno =-1"，then sendto always fail.What we do is resetting esp32 when we met "errno=-1" .What else can we do to make sendto success when we met "errno =-1"?

jixunx, 2017-11-02:
Thank you for your reply very much.
Once we met "errno==-1",then every "sendto" fails,and errno is -1.What we do is resetting esp32,what you do is close socket.Any other way can make "sendto" success after we met "errno ==-1"?