espressif/esp-idf #886: Malformed mgmt. frames in promiscuous mode

spacehuhn, 2017-08-07:
I'm working on a project using the promiscuous mode to sniff and save WiFi traffic.  
I noticed that all captured managment frames have 4 bytes at the end that don't make any sense.

That leads to malformed packets.
[image]

I'm using the Arduino IDE, however @Staubgeborener was able to reproduce the same issue using the ESP-IDF: https://gist.github.com/Staubgeborener/6977a6851510a296eabde0b9148f4658

I can provide more examples if necessary.
I bypassed this issue in my project with a simple if-statement: `if(type == WIFI_PKT_MGMT) len -= 4;`. So it doesn't save the last 4 bytes, if it's a managment frame.

SoftwareSamurai, 2017-10-10:
Isn't the last 4 bytes of any wifi packet the FCS? 
Or are you saying the packet length for management frames are including 4 extra bytes beyond the FCS?

Staubgeborener, 2017-10-10:
@SoftwareSamurai Those 4 bytes are appended after the checksum.

liuzfesp, 2018-01-15:
Hi @Staubgeborener @SoftwareSamurai I guess you got an CRC error packet. 

1. Check wifi_pkt_rx_ctrl_t.rx_state first before parsing the packet content. If wifi_pkt_rx_ctrl_t.rx_state is 0, it means the packet has no error, otherwise it has something wrong. 
2. The length of the packet contains 4 Bytes FCS, so for the packet without error, just ignore the last 4 bytes.

closed 2018-01-16 by FayeY

drws, 2021-08-18:
@spacehuhn and @Staubgeborener, have you been able to confirm @liuzfesp's prediction about frames aving CRC errors or is the same happening with CRC-successful frames? How did you solve the issue in your final code?

TianaESP, 2021-11-16:
@drws  Hi, can you please let us know what specific issue you met?

drws, 2021-11-18:
I only used basic promiscuous mode functions and was still researching more advanced topics when I stumbled upon this report. I thought it would be interesting to hear feedback from OP or other participants.

Staubgeborener, 2021-11-18:
I can’t say anything more about it, @drws . This issue was opened a few years ago and I'm currently neither working with the esp mcu nor with the promiscuous mode, so I can't tell you how it turned out back then. Maybe @spacehuhn can tell you something else.

spacehuhn, 2021-11-20:
I was able to fix the malformed packet error by implementing what @liuzfesp said. Not sure what else to tell you. ¯\_(ツ)_/¯

drws, 2021-11-20:
Thank you all for your input :)