espressif/esp-idf #887: Connection parameter conditions

jkhax0r, 2017-08-07:
Is this code on line 640 of l2c_ble.c a bug?
`
                    timeout <= ((1 + latency)*max_interval*2) ||
`

Based on the commented out (floating point code), 
`/*(timeout >= max_interval && latency > (timeout * 10/(max_interval * 1.25) - 1)) ||*/`

it seems the timeout comparison should be as follows:
`                   timeout <= ((1 + latency)*max_interval/8) ||
`

I have connection params of min=6 (7.5ms), max=6 (7.5ms), latency=499, conn=3000 (30000ms), which I believe is valid.   I was getting an error messages about invalid params, and traced it to this line of code. 

Some more info and further links to BLE spec connection timing requirements is here: https://devzone.nordicsemi.com/question/46510/how-can-the-peripheralslave-sleep-as-much-as-possible/

Weijian-Espressif, 2017-08-09:
hi @jkhax0r , This is a bug, we will fix the bug in the latest idf. Thanks for your reminder.
Here is the correct code:

if (min_interval < BTM_BLE_CONN_INT_MIN || min_interval > BTM_BLE_CONN_INT_MAX ||
                    max_interval < BTM_BLE_CONN_INT_MIN || max_interval > BTM_BLE_CONN_INT_MAX ||
                    latency  > BTM_BLE_CONN_LATENCY_MAX ||
                    timeout < BTM_BLE_CONN_SUP_TOUT_MIN || timeout > BTM_BLE_CONN_SUP_TOUT_MAX ||

                    /* The supervision_timeout parameter defines the link supervision timeout for the connection.
                       The supervision_timeout in milliseconds shall be large than (1 + latency) * max_interval * 2,
                       where max_interval is given in milliseconds. (See [Vol 6] Part B, Section 4.5.2).
                       supervision_timeout (mult of 10ms); conn_interval (mult of 1.25ms)
                       (max_interval * 1.25 * 2) replaced by ((max_interval * 5) >> 1).
                    */
                    ((timeout * 10) < ((1 + latency) *((max_interval * 5) >> 1))) ||
                    max_interval < min_interval) {
                l2cu_send_peer_ble_par_rsp (p_lcb, L2CAP_CFG_UNACCEPTABLE_PARAMS, id);
                UINT8 status = HCI_ERR_PARAM_OUT_OF_RANGE;
                l2c_send_update_conn_params_cb(p_lcb, status);

jkhax0r, 2017-08-10:
Thanks looks good

closed 2017-08-10 by jkhax0r