espressif/esp-idf #888: bta_gattc_register_service_change_notify() causes GATT_Write failures

jkhax0r, 2017-08-07:
I've tracked down a possible bug/regression that I believe is coming from commit: https://github.com/espressif/esp-idf/commit/c5debed62106e533b740cfae6458f2f55fba8853

If I connect a peripheral device to the ESP32, and then reset the peripheral, when it comes back up and reconnects, I can't write a characteristic anymore.  The first characteristic write I do silently fails with no callback - neither success nor failure.  Only a message in the logs that says "GATTC_Write: GATT_BUSY" (see below logs).  I tracked down that the stack is originally getting into that "busy state" from a GATTC_Write call initiated by the service change discovery code added by this commit.  When I remove all changes related to that commit, it works every time.  

Hopefully I'm not doing something stupid with my peripherals service change service, but best I can tell this is implemented in the peripheral's stack and I couldn't change any code here (it is the CC2540 chipset).

Logs below with GATT VERBOSE logging enabled. 

```
I (19446) ble_rmt: Opening gattc connection for gattc_if=4 to 20:c3:8f:ad:62:50
E (19447) BT: GATT_Connect gatt_if=4
E (19447) BT: gatt_get_ch_state: ch_state=0
E (19447) BT: gatt_set_ch_state: old=0 new=2
E (19449) BT: gatt_update_app_use_link_flag  is_add=1 chk_link=0
E (19449) BT: gatt_update_app_hold_link_status found=1[1-found] idx=0 gatt_if=4 is_add=1
E (19449) BT: gatt_get_ch_state: ch_state=2
E (19450) BT: GATT_GetConnIdIfConnected status=0

I (19450) ble_rmt: REG FOR_NOTIFY: status 0, srvc_id = 1d17, char_id = 1d18
E (19749) BT: GATT   ATT protocol channel with BDA: 20c38fad6250 is connected
E (19749) BT: gatt_is_bda_in_the_srv_chg_clt_list :20-c3-8f-ad-62-50
E (19750) BT: gatt_get_ch_state: ch_state=2
E (19750) BT: gatt_set_ch_state: old=2 new=4
E (19750) BT: gatt_connect_cback: from 20c38fad6250 connected:1 conn_id=1 reason = 0x0000
E (19751) BT: gatt_get_ch_state: ch_state=4
E (19751) BT: GATT_GetConnIdIfConnected status=1

E (19751) BT: JK: bta_gattc_conn_cback 0 0
E (19752) BT: gatt_num_apps_hold_link   num=1
E (19752) BT: GATT_SetIdleTimeout idle_tout=65535 status=1(1-OK 0-not performed)
E (19753) BT: GATT_GetConnectionInfor conn_id=4
E (19754) BT: l2cble_start_conn_update, the last connection update command still pending.
E (19755) BT: l2cble_start_conn_update, the last connection update command still pending.
E (19755) BT: GATTC_Discover conn_id=4 disc_type=1
I (19756) GATTS: ESP_GATTS_CONNECT_EVT, gatts_if 3, conn_id 0, remote 20:c3:8f:ad:62:50, is_conn 1
I (19757) ble_rmt: ESP_GATTC_CONNECT_EVT
I (19757) ble_rmt: ESP_GATTC_OPEN_EVT
E (19849) BT: gatt_get_ch_state: ch_state=4
E (19949) BT: gatt_get_ch_state: ch_state=4
E (19949) BT: gatt_end_operation status=0 op=1 subtype=1
E (19950) BT: GATTC_Discover conn_id=4 disc_type=3
E (20049) BT: gatt_get_ch_state: ch_state=4
E (20049) BT: gatt_process_error_rsp
E (20049) BT: gatt_proc_disc_error_rsp reason: 0a cmd_code 0008
E (20050) BT: Discovery completed
E (20050) BT: gatt_end_operation status=0 op=1 subtype=3
E (20050) BT: GATTC_Discover conn_id=4 disc_type=4
E (20099) BT: l2cble_start_conn_update, staus = 1, line = 509
E (20107) BT: gatt_get_ch_state: ch_state=4
E (20121) BT: gatt_get_ch_state: ch_state=4
E (20136) BT: gatt_get_ch_state: ch_state=4
E (20136) BT: gatt_process_error_rsp
E (20137) BT: gatt_proc_disc_error_rsp reason: 0a cmd_code 0008
E (20137) BT: Discovery completed
E (20137) BT: gatt_end_operation status=0 op=1 subtype=4
E (20137) BT: GATTC_Discover conn_id=4 disc_type=3
E (20151) BT: gatt_get_ch_state: ch_state=4
E (20152) BT: gatt_process_error_rsp
E (20152) BT: gatt_proc_disc_error_rsp reason: 0a cmd_code 0008
E (20152) BT: Discovery completed
E (20152) BT: gatt_end_operation status=0 op=1 subtype=3
E (20152) BT: GATTC_Discover conn_id=4 disc_type=4
E (20166) BT: gatt_get_ch_state: ch_state=4
E (20181) BT: gatt_get_ch_state: ch_state=4
E (20182) BT: gatt_process_error_rsp
E (20182) BT: gatt_proc_disc_error_rsp reason: 0a cmd_code 0008
E (20182) BT: Discovery completed
E (20182) BT: gatt_end_operation status=0 op=1 subtype=4
E (20182) BT: GATTC_Discover conn_id=4 disc_type=5
E (20196) BT: gatt_get_ch_state: ch_state=4
[... 126 lines cut ...]

I (48699) ble_rmt: REG FOR_NOTIFY: status 0, srvc_id = 1d17, char_id = 1d18
E (48999) BT: GATT   ATT protocol channel with BDA: 20c38fad6250 is connected
E (48999) BT: gatt_is_bda_in_the_srv_chg_clt_list :20-c3-8f-ad-62-50
E (49000) BT: gatt_get_ch_state: ch_state=2
E (49000) BT: gatt_set_ch_state: old=2 new=4
E (49000) BT: gatt_connect_cback: from 20c38fad6250 connected:1 conn_id=1 reason = 0x0000
E (49001) BT: gatt_get_ch_state: ch_state=4
E (49001) BT: GATT_GetConnIdIfConnected status=1

E (49003) BT: JK: bta_gattc_conn_cback 0 0
E (49003) BT: gatt_num_apps_hold_link   num=1
E (49003) BT: GATT_SetIdleTimeout idle_tout=65535 status=1(1-OK 0-not performed)
E (49003) BT: GATT_GetConnectionInfor conn_id=4
I (49004) GATTS: ESP_GATTS_CONNECT_EVT, gatts_if 3, conn_id 0, remote 20:c3:8f:ad:62:50, is_conn 1
I (49004) ble_rmt: ESP_GATTC_CONNECT_EVT
I (49004) ble_rmt: ESP_GATTC_OPEN_EVT
E (49005) BT: GATTC_Write GATT_BUSY conn_id = 4
E (49005) BT: No pending command
E (49099) BT: gatt_get_ch_state: ch_state=4
E (49099) BT: gatt_end_operation status=0 op=3 subtype=2
E (49100) BT: No pending command
I (123650) ota: Connect to Wifi ! Start to Connect to Server....

```

nkolban, 2017-08-09:
What build of the ESP-IDF are you using?  I'm suffering from an apparent regression at Master that works when I go back to the 2.1 release ... see #857

jkhax0r, 2017-08-13:
It's pretty easy to confirm.  Just go look in the commit I referenced and remove all the added code relating to the service change characteristic.  It's all the new code in the bat_gatc_act.c file.  I'm running that way now until this gets fixed.  

But unless you're getting that Gattc_Write error printed to the console, and the "No pending command" error, I doubt these are related.

Yulong-espressif, 2017-09-28:
@jkhax0r When the ble stack doing service change discovery, you can't sent the write request at that time, so it will case the GATTC BUSY. Now we added the command queue for it, so for the new SDK, will not case the GATTC BUSY even you sent many request at the same time, thanks.

closed 2017-10-16 by FayeY