espressif/esp-idf #891: ETH example + sd_card example cause flash SPI issues

Gfast2, 2017-08-09:
Hallo Guy.

Recently I'm tunning example codes from esp-idf. They are [ETH-example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet) and [sd_card-example](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card). Because they both are using relative many pins. I'm not sure what cause the problem. The error Code:

```
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.0-beta3
Connecting........_____....._____....._____....._____.....___
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 14128 bytes to 8291...
Wrote 14128 bytes (8291 compressed) at 0x00001000 in 0.1 seconds (effective 1177.9 kbit/s)...
Hash of data verified.
Compressed 380576 bytes to 187020...
Wrote 380576 bytes (187020 compressed) at 0x00010000 in 4.6 seconds (effective 655.2 kbit/s)...
File  md5: 254f3ca3bece7e8662784f6820e5e894
Flash md5: 1be197b2bb8dca47d9e0b03a3de4c6d1
MD5 of 0xFF is 8d3ab494e9f903db00865ccc240e6920

A fatal error occurred: MD5 of file does not match data in flash!
/home/gfast2/esp/esp-idf/components/esptool_py/Makefile.projbuild:52: recipe for target 'flash' failed
make: *** [flash] Error 2

```
Wiring of the test set is following the [Frank's flavor](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/). The difference is only I'm using GPIO16 as PHY_POWER controller. And I figure out no matter which project I'd try to flash I just got the same effect. But if I put the devboard (sparkFun Thing) off the PCB, It can be flashed as usual. 
Because I got hanged for about two days. I've got some unstanderation about `strapping pins`. In this point I'm not sure what I've done uncorrectly. Because I've try these examples separetely without problem. And All the extern pullup/down resistors are also added as in the examples descriped.
Any suggestions?

igrr, 2017-08-09:
Pullup resistor on GPIO12 causes the flash voltage bootstrapping pin to be high, selecting 1.8V flash voltage. You need to follow the instructions in https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12 and burn EFUSE to select 3.3V flash voltage. Then the pull-up on GPIO12 will not affect SPI flash.

Gfast2, 2017-08-09:
Thanks @igrr, I'll give it a try. And write you the result later!

Gfast2, 2017-08-09:
Hi @igrr ,
It works like a charm!! I really appreciate your helps!
I know what I'd ask here may/should ask goolge first, But I just can't stop to knowing them as efficient as possible:
- Is that means, That sparkFun are using the interlnal regulator?
- How can I know this through Hardware / Software?
- Why if I only test the sd_card example won't expirience this issue?
 - Is that means, I can deactivate all strapping pins functions by manually setting up all ESP32 natures using espefuse tool? (All features descripted in [ESP32_datasheet](https://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) in Table 3)

igrr, 2017-08-09:
> Is that means, That sparkFun are using the interlnal regulator?
How can I know this through Hardware / Software?

Check the board schematics and see what is connected to VDD_SDIO pin of ESP32. If it is only connected to the flash chip and decoupling capacitor, then it means that internal regulator within ESP32 is used to power the flash chip. If the net also connects to VDD (power supply), then it means that the flash is powered directly from the power supply and the internal regulator of ESP32 is not used (disabled).

> Why if I only test the sd_card example won't expirience this issue?

No idea, frankly. You should see this even without Ethernet. There is a slight chance that flash can still work at lower voltage (1.8V) even if it rated for 3.3V, so maybe that was the case.

> Is that means, I can deactivate all strapping pins functions by manually setting up all ESP32 natures using espefuse tool? (All features descripted in ESP32_datasheet in Table 3)

Not all. You can control VDD_SDIO voltage, and that's about it. Boot mode, log output are still controlled by bootstrapping pins only.

Gfast2, 2017-08-09:
Hi @igrr ,
Thanks again. I think this should be the proper time to close this quick issue.
Interestingly, the ESP32 sparkFun Thing is not using the internal regulator at all. (Its [Schecmatic](https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/esp32-thing-schematic.pdf) is showing very clear that GPIO26 (VDD3P3_SDIO) is routed to a "jumper" which let the user to select how to power the W25Q32FVSS SPI Flash (Reason: Tinker can mod the SPI with a 16MB one? ;D) ). 3.3V from power supply is set as default power source for the SPI Flash. 
I suspect the setting we are talking all the time here can set up also how to read the SPI (1.8V or 3.3V), which cause the interesting issue. ;)
Here is the snapshot of its schecmatic:
[image]

closed 2017-08-09 by Gfast2