espressif/esp-idf #892: Can I package all SD card control in one freeRTOS task?

Gfast2, 2017-08-09:
Hi esp-idf,

I clearly know the most problem I knew is caused by the thing between my ears. But I still collected very limited ways to deal with bugs like this:
As I mentioned today morning. My test set is a merge of [Ethernet example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet) and [sd_card example](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card). The hardware wiring take [Frank's flavor](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/) for ETH, the sd_card wiring is just as it is from its README. The single difference is I'm using GPIO16 to drive PHY_POWER.
Right now I'm experencing a interesting effect:
If I put all stock code from sd_card example into a freeRTOS `task`, I'd throw the following error (crash):
```
E (246) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109
E (256) SD_CARD: Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400dec42  PS      : 0x00060530  A0      : 0x00000000  A1      : 0x3ffcda30  
0x400dec42: sd_card_task at /home/gfast2/workspace/ioc-110-ethernet/main/./sd_card.c:50 (discriminator 1)

A2      : 0x00000109  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800dec42  A9      : 0x3ffcd9e0  
A10     : 0x400d5760  A11     : 0x3ffb23b4  A12     : 0x3f403564  A13     : 0x0000001f  
0x400d5760: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:35

A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400dec42:0x3ffcda30 0x00000000:0x3ffcdb10
0x400dec42: sd_card_task at /home/gfast2/workspace/ioc-110-ethernet/main/./sd_card.c:50 (discriminator 1)

Entering gdb stub now.
$T04#b8GNU gdb (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/gfast2/workspace/ioc-110-ethernet/build/ethernet_demo.elf...done.
Remote debugging using /dev/ttyUSB0
0x400dec42 in sd_card_task (pvParameters=<optimized out>) at /home/gfast2/workspace/ioc-110-ethernet/main/./sd_card.c:50
50				ESP_LOGE(TAG, "Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
(gdb) l
45		esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);
46		if (ret != ESP_OK) {
47			if (ret == ESP_FAIL) {
48				ESP_LOGE(TAG, "Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
49			} else {
50				ESP_LOGE(TAG, "Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
51			}
52			return;
53		}
54	
(gdb) 

```

The `ERROR 0x109` is `ESP_ERR_INVALID_CRC`.
If I move all the sd_card code into app_main() in `main.c`, then the other part of the story - ETH won't works (doesn't crash):
```
I (1015) cpu_start: Pro cpu up.
I (1027) cpu_start: Starting app cpu, entry point is 0x40080df8
0x40080df8: call_start_cpu1 at /home/gfast2/esp/esp-idf/components/esp32/./cpu_start.c:174

I (0) cpu_start: App cpu up.
I (1060) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1082) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1102) heap_alloc_caps: At 3FFBB850 len 000247B0 (145 KiB): DRAM
I (1124) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1145) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1167) heap_alloc_caps: At 40088148 len 00017EB8 (95 KiB): IRAM
I (1187) cpu_start: Pro cpu start user code
I (1247) cpu_start: Starting scheduler on PRO CPU.
I (1250) cpu_start: Starting scheduler on APP CPU.
I (1250) eth_example: Initializing SD card
Name: 1232 
Type: SDSC
Speed: default speed
Size: 1932MB
CSD: ver=0, sector_size=512, capacity=3956736 read_bl_len=10
SCR: sd_spec=2, bus_width=5
I (1270) eth_example: Opening file
I (1300) eth_example: File written
I (1310) eth_example: Renaming file
I (1310) eth_example: Reading file
I (1320) eth_example: Read from file: 'Hello 1232 !'
I (1320) eth_example: Card unmounted
I (1320) eth_example: sd_card job done!
I (1370) emac: emac start !!!

I (1380) emac: emac resetting ....
I (1380) emac: emac reset done
I (1390) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1400) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
E (2400) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (3400) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (4400) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (5400) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: btn_task
E (6400) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (7400) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (8400) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
E (9400) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff
E (10400) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: btn_task

```
From these errors I can tell, it must have something to do with things like strapping pins or like so.
Interestingly I found out, ETH  is not allowed to be packed into its own task neither. How can I understand this problem? Thanks for the advice.

Gfast2, 2017-08-09:
I traced down what does `handle_idle_state_events` do. in `esp-idf/components/driver/sdmmc_private.h` I found a structure that define the possible events for sdmmc:
```
typedef struct {
    uint32_t sdmmc_status;      ///< masked SDMMC interrupt status
    uint32_t dma_status;        ///< masked DMA interrupt status
} sdmmc_event_t;
```
According to this trace output:
`E (10485) sdmmc_req: handle_idle_state_events unhandled: 00008000 00000000`
and the definition of the function which throw this output (from `esp-idf/components/driver/sdmmc_transaction.c`):
```
static esp_err_t handle_idle_state_events()
{
    /* Handle any events which have happened in between transfers.
     * Under current assumptions (no SDIO support) only card detect events
     * can happen in the idle state.
     */
    sdmmc_event_t evt;
    while (sdmmc_host_wait_for_event(0, &evt) == ESP_OK) {
        if (evt.sdmmc_status & SDMMC_INTMASK_CD) {
            ESP_LOGV(TAG, "card detect event");
            evt.sdmmc_status &= ~SDMMC_INTMASK_CD;
        }
        if (evt.sdmmc_status != 0 || evt.dma_status != 0) {
            ESP_LOGE(TAG, "handle_idle_state_events unhandled: %08x %08x",
                    evt.sdmmc_status, evt.dma_status);
        }

    }
    return ESP_OK;
}
```
Now I can tell a step forward: When the sd_card example code is placed into a freeRTOS task, it will experience a event that I (the example code) still not handled -> a `card detection event` ! 💃 

Because the lack of related infos, I still not fix this issue, And I event don't understand why there is such a event (The card detection pin is not defined at all.). If I got something, I will document it here.

negativekelvin, 2017-08-10:
1. Your SD task has to run forever or end with vtaskdelete
2. 00008000 means end bit error not card detect
3. You changed CONFIG_PHY_POWER_PIN to 17?

Gfast2, 2017-08-10:
>Your SD task has to run forever or end with vtaskdelete

I've tried to terminate `sd_card_task` with `vTaskDelete(NULL);`. Till now I don't think too much about this.

>00008000 means end bit error not card detect. I don't know why it happens if you didn't change anything. 

Wow, this should be a very useful tip for me. I'll try to figure out how can I find such info in source code, but if you have the time. Can you tell me how did you figure out specifically this errors meaning? (Yesterday I did investigate about 1 h try to intepretate this)

>Timestamps are weird.

Yes, I will try it today instantly, If I got these Timestamps another time. I will leave info here.

>You changed CONFIG_PHY_POWER_PIN to 16?

Yes. I may cause issue, because I do use `UART1` for reading GPS signal, too. BTW, can init `UART1` without using its `rx` pin (`GPIO16`)?

Gfast2, 2017-08-10:
Hi esp-idf,

I belive I'm doing some intertask communication wrong & need to investigate more time on the differences between `xTaskCreatePinnedToCore()` and `xTaskCreate()`

---
Right now I've did the following test set:
if I let the whole example code of sd_card in `app_main()`, and init my `button_task` like this
`xTaskCreatePinnedToCore(&button_task, "btn_task", 20000, NULL, 10, NULL, 1);`
Then ETH Task will works fine, but the button_task is somehow works (I will trigger events which will read the button change status and send it to LED task, which still get these event correctly), but trigger its task watchdog:
```
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: btn_task
```
if I leave the whole test set not changed at all except that I init my `button_task` in another way:
`xTaskCreate(&button_task, "btn_task", 20000, NULL, 10, NULL);`
Then I got everything start correctly, except that the ETH task not start at all.

BTW: In both situation, all sd_card example code works just fine at the very beginning.

---
Though before the integration of sd_card control I've got this warning during compiling time:
`/home/gfast2/esp/esp-idf/components/freertos/include/freertos/queue.h:425:115: warning: passing argument 3 of 'xQueueGenericSend' makes integer from pointer without a cast [-Wint-conversion]`
Perhapse that is the real problem. I'll try it shortly. 
I don't experience any wierd Timestamp issue any more by far.

negativekelvin, 2017-08-10:
OK well you probably have some loop in your task.  You should try the forum for coding help.  Github issues is for bug reports.

Gfast2, 2017-08-10:
@negativekelvin ,

Thanks again for your advices. 🉐 
The problem is get fixed by adding a 1.5 sec. delay (`vTaskDelay(1500 / portTICK_PERIOD_MS)`) in `app_main()` directely after the init of `sd_card_task`. 😃 💡 

---

The think after this fix is: Is that saying the `sdmmc` related driver and routines are `thread/task` sensitive?

But any way, it works after all. I'm happy with the result firstly. If I need a much performanter flavor, I'd open another issues. (Sure, I will investigate many hours before this)

>You should try the forum for coding help. Github issues is for bug reports.

Yeh, I know here is only for _issues_. But to tell the truth, I came from the era without much experience / practice with forum. But I will keeping on trying this!

Tanks & have a nice day. 🥇

closed 2017-08-10 by Gfast2