espressif/esp-idf #893: Allow Port 0 in mdns

johsely1234, 2017-08-09:
Currently it is not possible to set Port 0 in 
`esp_err_t mdns_service_add(mdns_server_t * server, const char * service, const char * proto, uint16_t port)`
because of the check 
if (!server || !service || !proto || **!port**) {

in the file mdns/mdns.c

Could you please just remove the !port so that I can use Port 0.
Best regards

me-no-dev, 2017-08-14:
Port 0? What is the use case here?

me-no-dev, 2017-08-14:
https://www.lifewire.com/port-0-in-tcp-and-udp-818145

johsely1234, 2017-08-14:
The Mdns advertising should show port 0. I don't think the connection then really uses this port.

But this is defined by someone else, I can't change it.

johsely1234, 2017-08-14:
But if it really makes no sense then I have to use another port anyway.

negativekelvin, 2017-08-14:
Which "someone else" looks for a service advertised on port 0?

johsely1234, 2017-08-16:
The client part of the connection is made by another person. In their old system they use port 0, so they want to use port 0 also in the new system. But I'll try to convince them not to use port 0.

closed 2017-08-16 by johsely1234