espressif/esp-idf #894: [Request] Make it possible to change IP Adress of Access Point

johsely1234, 2017-08-09:
Currently I could not find a way to set the IP-Adress of the Access Point of the esp32.

That would be a very nice feature!

Best regards

liuzfesp, 2017-08-15:
HI @johsely1234 ,
For DHCP Server:
tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_SET, TCPIP_ADAPTER_REQUESTED_IP_ADDRESS, &lease, sizeof(lease)); //set IP
tcpip_adapter_dhcps_option(TCPIP_ADAPTER_OP_SET, TCPIP_ADAPTER_IP_ADDRESS_LEASE_TIME, &lease_time, sizeof(lease_time)); //set lease time
...

For static IP:
tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP); //stop dhcp server firstly
tcpip_adapter_set_ip_info(WIFI_IF_AP, &ip); //set static IP

everslick, 2017-08-15:
The Arduino way is to use: WiFi.softAPConfig(addr, gateway, netmask);

johsely1234, 2017-08-16:
I will try and close.
I tried to changed it in the ip settings but that did not work.

johsely1234, 2017-08-17:
Ok thats what I wanted and so it worked:

```
tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);

tcpip_adapter_ip_info_t ip_info;
IP4_ADDR(&ip_info.ip,192,168,1,1);
IP4_ADDR(&ip_info.gw,192,168,1,1);
IP4_ADDR(&ip_info.netmask,255,255,255,0);
printf("set ip ret: %d\n", tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &ip_info)); //set static IP

tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP);
```

Thanks a lot!

closed 2017-08-17 by johsely1234