espressif/esp-idf #898: ota data partition invalid and no factory, can't boot

hhcc2017, 2017-08-10:
I have downloaded a wrong file on the address 0x8000 and ESP32 can't boot anymore. Below is the log. I think I have erased the nvs data.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4644
load:0x40078000,len:11656
load:0x40080000,len:252
entry 0x40080034
[0;32mI (47) boot: ESP-IDF v1.0-1566-g9256b7c 2nd stage bootloader[0m
[0;32mI (47) boot: compile time 20:57:17[0m
[0;32mI (127) boot: Enabling RNG early entropy source...[0m
[0;32mI (127) boot: SPI Flash RID  : 0x1640EF[0m
[0;32mI (140) boot: SPI Flash  MF  : 0xEF[0m
[0;32mI (179) boot: SPI Flash  ID  : 0x4016[0m
[0;32mI (219) boot: SPI Speed      : 40MHz[0m
[0;32mI (258) boot: SPI Mode       : DIO[0m
[0;32mI (296) boot: SPI Flash Size : 4MB[0m
[0;32mI (334) boot: Partition Table:[0m
[0;32mI (368) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (436) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (506) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (575) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (645) boot:  3 miio_fw1         OTA app          00 10 00010000 00160000[0m
[0;32mI (715) boot:  4 miio_fw2         OTA app          00 11 00170000 00160000[0m
[0;32mI (785) boot:  5 test             test app         00 20 002d0000 00013000[0m
[0;32mI (854) boot:  6 mimcu            Unknown data     01 fd 002e3000 00100000[0m
[0;32mI (924) boot:  7 coredump         Unknown data     01 03 003e3000 00010000[0m
[0;32mI (994) boot:  8 minvs            Unknown data     01 fe 003f8000 00004000[0m
[0;32mI (1064) boot: End of partition table[0m
[0;31mE (1104) boot: ota data partition invalid and no factory, can't boot[0m
user code done

igrr, 2017-08-10:
You need to flash the partition table again. You can do this by running
'make flash'.

FayeY, 2017-08-14:
Hi @hhcc2017 , did the above suggestion help you resolve the issue?

projectgus, 2017-08-14:
In addition to @igrr 's suggestion, a proper fix for this problem is coming in IDF V3.0 (it will be merged to master shortly), meaning that invalid OTA data will no longer halt the boot (the bootloader will fall back to a sensible choice, in this case it will search starting from OTA 0.)

projectgus, 2017-08-17:
This fix is now in the IDF master branch, as of 969f1bb9be1.

closed 2017-08-17 by projectgus