espressif/esp-idf #899: SD: Formatting issue with 1-line mode, SPI works fine

Staubgeborener, 2017-08-10:
I noticed the following: When i try to save some dump (generated by network traffic / promiscuous mode) in 1-line mode to my sd card, the format is broken. Looks like this:
```
000068 02 0C 00 0B 05 02 00 0C 
000070 00 00 46 05 32 08 01 00 
00007878 00 2D 1A BC 09 1B FF FF 

0000D8 D4 6C 38 00 ED FF000000 
80 00 00 00F F FF FF FF 
000008 FF FF 8C 34 FD 31 08 9F 
```
Just as an example. Some pakets are 'normal' and some others looks like this. Maybe a paragraph is missing, maybe a space, maybe there are 3 chars printed, etc.

So i tried the sd card in SPI. And here is everythings fine. My code is [here](https://gist.github.com/Staubgeborener/379a494bfd11dd671873468d2b0947aa).

Basically i put the buffer from the promiscuous callback into a ringbuffer and fomat the buffer in a nice hexdump with the following code

```c++
for (int i = 0; i < packet->rx_ctrl.sig_len; i++) {
    if (i % 8 == 0) {
        //Offset hinzufügen
        snprintf(hexbuf, sizeof(hexbuf), "%06X ", offset);
        strcat(hexbuf_dump, hexbuf);
        offset += 8;
    }
    //Hex anfügen
    snprintf(hexbuf, sizeof(hexbuf), "%02X ", packet->payload[i]);
    strcat(hexbuf_dump, hexbuf);
    //Absatz, wenn: 8x Hex oder Frame zuende
    if ((i % 8 == 7) || ((i + 1) == packet->rx_ctrl.sig_len)) {
        snprintf(hexbuf, sizeof(hexbuf), "\n");
        strcat(hexbuf_dump, hexbuf);
    }
}
sprintf(filenamebuf, "/sdcard/%s.pcap", filename);
printf("%s", hexbuf_dump);
savestuff(hexbuf_dump);
memset(&hexbuf_dump[0], 0, sizeof(hexbuf_dump));
offset = 0;
```

saving function is here
```c++
void savestuff(const char* data)
{
    //fprintf("%c", data);
    sprintf(filenamebuf, "/sdcard/%s.pcap", filename);
    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen(filenamebuf, "a");

    if (f == NULL) {
        ESP_LOGE(TAG, "Fehler! Datei konnte nicht geoeffnet werden!");
        digitalWrite(led1, HIGH);
        vTaskDelay(500 / portTICK_PERIOD_MS);
        digitalWrite(led1, LOW);
        ESP.restart();
    }
    ESP_LOGI(TAG, "File open");
    //fprintf(f, "%s", data);
    fwrite(data, sizeof(char), strlen(data), f);
    fclose(f);
    ESP_LOGI(TAG, "File written");

}
```

igrr, 2017-08-10:
One of the issues you may be seeing, where some characters appear doubled, has been recently fixed in b260f6cb25bf7ab9be907fe4fdebaac7aacb8900. Please try updating IDF and check if it works any better.

Staubgeborener, 2017-08-10:
I updated the IDF (https://github.com/espressif/esp-idf/releases -> Obtaining v2.1 -> using git), but in the new folder 'esp-idf-v2.1' are still the old files (i have compared the content of those files with your updated versions). Also in the .zip file are only the old files.

Staubgeborener, 2017-08-10:
Well, thanks. That works perfectly.

closed 2017-08-10 by Staubgeborener