espressif/esp-idf #900: The reason of WIFI_PROTOCOL_LR

lixx5427, 2017-08-10:
您好，我现在需要使用ESP32的长距离通信功能，WIFI_PROTOCOL_LR设置和通信是好使的，但是感觉上通信距离并没有发生改变，我观察了使用WIFI_PROTOCOL_LR以后的通信的数据包的频谱分布，也没有发现和其它设置如WIFI_PROTOCOL_11B/11G有显著区别。

请问ESP32到底是因为什么实现了长距离通信，如何实地测试？

非常感谢

liuzfesp, 2017-08-15:
你好，如果要测试long range的话，可以将esp32 AP与esp32 sta同时设置成WIFI_PROTOCOL_LR only模式，再测试，具体实现方式公司内部机密，不便于透露:)，已有专利保护。

negativekelvin, 2017-08-15:
[This?](http://211.157.104.77:8080/sipo_EN/search/detail.do?method=view&parm=16b414c21a2f19d11b2c18401bcd1a5f183a19661ad91be11a501c58047927231f5c21a222572195236c20482755275723ca24be2221222525702494250d26c0274025822c3f29092a7c29a02d6d2d6f28fa2ed62bd12c892f482c34330947932f5c2c0a2ac731b9333c316c366534e7318235ee337934f1360837f03747371b371a34e231bf38f13b04390c3e0d3f6f39ea3dc63c573d6d3e683e4c26d918c33ffe3c523c273e354334405c47c545774302408e4423458147b04698463346aa46cc46da436f47554bcc49784a2d)

YuanChenmang, 2017-08-15:
I personally guess it should be analog LoRa similar to the modulation method,
Sacrifice bandwidth in exchange for improved signal to noise ratio.

lixx5427, 2017-08-15:
我做到实验，看过发射出来的频谱图，频带宽度并没有收缩，所以感到很迷惑.......

liuzfesp, 2017-10-10:
@lixx5427 我怀疑你LoRA only的功能没有跑起来，有几个条件可用于判断LoRA only是否配置成功： 1> 使用手机连配置了LoRA only 的AP，连不上 2> 使用配置了LoRA only 的设备去连路由器，连不上，3> 使用配置了LoRA only的sta连配置了LoRA only的AP,能够连上  如果1>2>3>都符合，说明LoRA配置正确。也可以通过get API获取一下当前的phy mode，如果为WIFI_PROTOCOL_LR only说明配置成功。 另外，最好能够将你的配置方法，测试环境，测试步骤，测试结果可以介绍一下。

closed 2017-11-21 by FayeY