espressif/esp-idf #901: start the commands "xtensa-esp32-elf-gdb -x gdbinit build/blink.elf" and the openocd reports the error below

zjykymt, 2017-08-11:
Error: 989 14792 esp108_common.c:177 esp108_do_checkdsr(): cpu0: esp32_fetch_all_regs (line 234): DSR (8011CC13) indicates DIR instruction generated an exception!

Error: 1007 14804 FreeRTOS.c:310 FreeRTOS_update_threads(): FreeRTOS maximum used priority is unreasonably big, not proceeding: 1061164244

I use the Jlink, and I can not debug the program. 
Hope to get your help.

Include two log files
[openocd.txt](https://github.com/espressif/esp-idf/files/1217332/openocd.txt)
[gdb_log.txt](https://github.com/espressif/esp-idf/files/1217334/gdb_log.txt)

zjykymt, 2017-08-11:
There is something worth sharing is that I have just succeeded in debugging.

The solution is that I am modifying "adapter_khz" parameter within "esp-wroom-32.cfg":
old:adapter_khz 20000
new:adapter_khz 2000

JTAG Wiring Table(JLINK same applies)
[image]

closed 2017-08-11 by zjykymt

jkhax0r, 2017-08-12:
Just an FYI, I have succeeded to use debugger with a wroom-32 based custom
hardware and the tiao usb jtag debugger at 20Mhz (and 26Mhz).

Thanks for the openocd support!

It may be that your wires are too long for 20mhz.  I even got it at 26mhz
but didn't want to push the limits too much and didn't see much performance
difference.  The jlink may be loading down the wires with too much cap or
series r.

Closed #901 <https://github.com/espressif/esp-idf/issues/901>.

<https://github.com/espressif/esp-idf/issues/901#event-1202278922>, or mute
the thread