espressif/esp-idf #902: Is Zephyr good enough for ESP32 & Is is FreeRTOS royalty free for commercial applications?

Gfast2, 2017-08-11:
Hi EPS32ers,

_I did contemplated this for a moment, And still very eager to know..._

I've hanging around & figured out Linux Foundation are pushing its [_**Zephyr**_ (RTOS)](https://www.zephyrproject.org/) as another _**Linux**_ but aiming at very resource constrain Embedded Hardwares. I digged even deeper, and discovered @igrr has a [forked](https://github.com/igrr/zephyr/) on it. I even found documentation on how to [port](https://www.zephyrproject.org/doc/boards/xtensa/esp32/doc/esp32.html) Zephyr on ESP32 directly.
Meanwhile I've seen some [disscussions](https://esp32.com/viewtopic.php?f=2&t=300&sid=3f4b49bc56ac81498efaa9415df53d45) on ESP32's Forum, too. But still not that in depth yet.

---

But I **still not** get the working status of _Zephyr_ porting on ESP32 Plattform. 
From its introduction, I've to say, its really really impressive:
>The Zephyr kernel is derived from Wind River's commercial VxWorks Microkernel Profile for VxWorks. Microkernel Profile has evolved over 20 years from DSP RTOS technology known as Virtuoso. The RTOS has been used in several commercial applications including satellites, military command and control communications, radar, telecommunications and image processing. The most recent example of the technology’s success is the successful Philae Landing on Comet Churyumov–Gerasimenko and the accompanying Rosetta Orbiter.

And What I really care is that it is really _free_ (commertial-neutral) 🤑 . It has to be the way of the future .🌌 

---

But as always, every selfrespect programmer has to try their best to live with the reality in peace ☮️ 
Because I'm planning to lifting my hobby projects based on ESP32 to commercial use somewhen later. I'd know
_**How much free I can get from FreeRTOS. & How to pay for the parts which is not free?**_
(ps: I know FreeRTOS is [GPL](https://www.gnu.org/licenses/gpl-3.0.de.html) Licensed, and what it means in very rough manner 🏷 )

negativekelvin, 2017-08-12:
Zephyr is nowhere near usable yet.  Hopefully later.

Freertos is modified gpl all you have to do is provide esp-idf source code but your app can be closed source.

Gfast2, 2017-08-14:
Thanks for @negativekelvin's answer! It means at least I can do something without Software Licencing problem in near futur firstly. 😀 
Thanks for @FayeY 's name correction. means "_**[TW#14723]**_" something about twitter ( If this question is too dummy🎋, ignore it. If not, please crash course me how can I find it in twitter. 💇 )

---

I surely know these kind of question shouldn't exist here (rather then in the [ESP32 Forum](https://esp32.com/) ). If you think I'm right. should I close this _"issue"_ here now? (I don't mind if you close it, if you think this diskussion is enough. ☯️ )

negativekelvin, 2017-08-14:
No not twitter, espressif's internal tracking reference with **no public visibility** 😠

projectgus, 2017-08-14:
Hi @Gfast2,

> Freertos is modified gpl all you have to do is provide esp-idf source code but your app can be closed source.

To be precise, most of ESP-IDF is under the Apache License or a compatible permissive license so there is no requirement to provide this source code (or pay anyone money). The only part which is not is the FreeRTOS kernel itself (under components/freertos). You need to distribute only those source files to meet the license requirements (the "FreeRTOS exception" added to the GPL license provides for this).

> I surely know these kind of question shouldn't exist here (rather then in the ESP32 Forum ). 

That's correct, esp32.com is a better place for this kind of discussion. Please feel free to post more questions there!

Angus

closed 2017-08-14 by projectgus