espressif/esp-idf #904: how to develop directly by registers not by APIs

TaoStarlit, 2017-08-12:
Hi, Ivan Grokhotkov,
In my project, I have to develop directly by registers, to get a high performance of ADC.
But in the esp-idf framework, the adc example and adc.h headerfile only have several functions, and even for the adc bit width configuration, they use the api: adc1_config_width (there are no function for adc2), besides, we can not go the function implement to learn how the operate the registers of adc1, that results in we we don't know how to operate adc2, as there are no api for adc2.
Thus, my point is how to read the libdriver source to see how to develop our preject directly by the register not by the API.

Sincerely,
Tao

negativekelvin, 2017-08-12:
https://github.com/espressif/esp-idf/issues/461

igrr, 2017-08-14:
ESP32 peripherals are documented in the ESP32 Technical Reference Manual. If you see that the documentation for some registers or some peripherals is missing or incomplete, please voice your concern in this forum topic: https://esp32.com/viewtopic.php?f=5&t=905. That is the best way to convey information to the documentation team.

On the topic of ADC2: if you are using WiFi, then ADC2 is not available to applications until an API for this purpose is added to ESP-IDF. The reason is that ADC2 is also used by WiFi driver, and the application can only use ADC2 when WiFi driver is not using it (and is not about to use it). This coordination mechanism is work in progress at the moment.

If you are not using WiFi in your application, then ADC2 can probably be controlled with the code in PR which @negativekelvin has linked above.

closed 2017-08-14 by igrr