espressif/esp-idf #908: BLE: Failing to esp_ble_gattc_open a device without scanning first

rojer, 2017-08-14:
`esp_ble_gattc_open` fails if the device is using random address.
for example, [nRF51822 Smart Beacon](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF51822-Bluetooth-Smart-Beacon-Kit).
with extra debug enabled, i see this in the log:
```
[Aug 14 13:24:30.465] E (2530768) BT: BTM_InqDbRead: bd addr [eb12dd51193d]
[Aug 14 13:24:30.474] E (2530778) BT: btm_find_dev_type - unknown device, BR/EDR assumed
[Aug 14 13:24:30.484] E (2530788) BT: btm_find_dev_type - device_type = 1 addr_type = 0
```
well, the guess is wrong: it's a BLE device (type 2) and the correct address type should be 1 (`BLE_ADDR_RANDOM`).
this works if prior to connecting a scan is performed and BTM has a chance to populate its inquiry DB so that this information is available when `esp_ble_gattc_open` is invoked.
however, the documentation does not mention this as a requirement.
so please clarify: is this the intended behavior and users should always perform a scan before connecting or if it is a bug and `esp_ble_gattc_open` should perform its own scan if necessary?

FayeY, 2017-09-26:
Hi @rojer , this issue should have been fixed in the latest IDF, please have a try.

rojer, 2017-09-26:
i'd love to, but recent changes broke GATTC API and made it even harder to use. my code doesn't compile and my brain hurts trying to figure out the new way. i filed https://github.com/espressif/esp-idf/issues/1041 to get some help.

closed 2017-10-10 by FayeY

rojer, 2017-10-18:
confirmed as fixed