espressif/esp-idf #911: Stop scanning after few time

esantosjr, 2017-08-15:
Hi,

I am developing a project using ESP32 with the gattc_demo example that must scan for beacons and print its parameters as MAC address. I was able to run and see some information, but I have to scan continuously, something like **esp_ble_gap_start_scanning(forever)**.

What I have done to implement this logic was recall the esp_ble_gap_start_scanning method inside the **ESP_GAP_SEARCH_INQ_CMPL_EVT** case to restart scanning. However, after some time, I can see only the message **BT: btc_search_callback  BLE observe complete. Num Resp 80** and no information about the beacons is printed. It seems like the scan was made but no devices were found.

I need some help to figure out what is going on. Is there a best way to implement this?

Thanks.

TianHao-Espressif, 2017-08-17:
@esantosjr , you mean ESP32 cannot scan any devices or just a certain device? Then, could the GATTC-DEMO in esp-idf/examples/bluetooth can scan the devices?

esantosjr, 2017-08-18:
Hi,

the ESP32 cannot scan any devices!
I already am using the GATTC-DEMO example with the modification of scan continuously.

This is an image of the result that I am getting. It seems that the scan is making "white reads".

[image]

hiik3n, 2017-08-23:
@esantosjr, @TianHao-Espressif  I also met this issue. 
**esp_ble_gap_start_scanning()** is used to start ble scan, and 
**E (947683) BT: btc_search_callback  BLE observe complete. Num Resp 211**
for indicating the end of scan duration. (I set duration = 10s)

There are some ble data captured during scan duration.However, after a while, there is no scanned ble data. After I restart esp32, it can capture some ble datas then nothing....

At first I followed the ESP_EDF BLE example, then I found this tutorial https://github.com/barnybug/ESP32BleMQTT
It said that:
 // scan params must be reset before starting another scan
            printf("scan completed, restarting...\n");
            esp_ble_gap_set_scan_params(&ble_scan_params);

I tried to call **esp_ble_gap_set_scan_params** everytime before start new scan duration.
But the issue still happens.

One thing is that, this issue happen for some esp32 boards (2/5 of my boards have issue)

Hope someone can give some advice.
Thank you

hiik3n, 2017-08-23:
@TianHao-Espressif is there any method to tracking or monitor the stack is used by any task (ble task for example)

TianHao-Espressif, 2017-08-23:
@hiik3n , please undefine the macro `CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE` by unselect menuconfig -> Component config -> PHY(or Wifi->PHY) -> Do phy calibration and store calibration data in NVS to do full RF calibration.
Then try it. Thanks.

hiik3n, 2017-08-23:
@TianHao-Espressif When I tried to disable CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE, the stituation is worse, onle first scan duratio has some response, others have nothing....

TianHao-Espressif, 2017-08-23:
@hiik3n , another suggestion: please do a flash erase by `make erase_flash`. To clear all the init_data and program.

BTW, so you mean that the same program burned into different ESP32 board, some ESP32 board can scan successfully, but others cannot scan anythings. Right?

hiik3n, 2017-08-23:
@TianHao-Espressif I also try make erase_flash, but no change.

I tried same program on 5 boards, 2 boards almost have issue after 1-3 hour of runing, others can stand for a day...

TianHao-Espressif, 2017-08-23:
@hiik3n , when the ESP32 board will stop?  After duration or in duration? I mean that if duration is 10s, after 10s, scan stop automatically, then do scan again, cannot scan anything, right?

hiik3n, 2017-08-23:
@TianHao-Espressif " duration is 10s, after 10s, scan stop automatically, then do scan again, cannot scan anything" >> yes

esantosjr, 2017-08-23:
@hiik3n and @TianHao-Espressif, I have got a solution for my problem.

First of all I removed all unused code for me, because I only want to scan devices, and not connect to one.

The other changes are commented in the code.

The scan ran for 3 days without any problems.

The resulting code is:

```c
/****************************************************************************
*
* This file is for gatt client. It can scan ble device.
*
****************************************************************************/

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"
#include "controller.h"

#include "bt.h"
#include "bt_trace.h"
#include "bt_types.h"
#include "btm_api.h"
#include "bta_api.h"
#include "bta_gatt_api.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"

#define GATTC_TAG "BEACON SCAN"
// Declare static functions
static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param);

static const char device_name[] = "Alert Notification";

static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_PASSIVE, // change scan type to passive
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x10,
    .scan_window            = 0x08
};

static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{	
    switch (event) {
    case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT: {
        // the unit of the duration is second
        uint32_t duration = 10;
        esp_ble_gap_start_scanning(duration);
        break;
    }
	
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT: 
			esp_log_buffer_hex(GATTC_TAG, scan_result->scan_rst.bda, 6);
            ESP_LOGI(GATTC_TAG, "Searched Adv Data Len %d, Scan Response Len %d", scan_result->scan_rst.adv_data_len, scan_result->scan_rst.scan_rsp_len);
            adv_name = esp_ble_resolve_adv_data(scan_result->scan_rst.ble_adv,
                                                ESP_BLE_AD_TYPE_NAME_CMPL, &adv_name_len);
            ESP_LOGI(GATTC_TAG, "Searched Device Name Len %d", adv_name_len);
            esp_log_buffer_char(GATTC_TAG, adv_name, adv_name_len);
            ESP_LOGI(GATTC_TAG, "\n");
            if (adv_name != NULL) {
                if (strlen(device_name) == adv_name_len && strncmp((char *)adv_name, device_name, adv_name_len) == 0) {
                    ESP_LOGI(GATTC_TAG, "Searched device %s\n", device_name);
                }
            }
            break;
        case ESP_GAP_SEARCH_INQ_CMPL_EVT:
			esp_bt_controller_disable(ESP_BT_MODE_BTDM); // reinit ... 
			esp_bt_controller_enable(ESP_BT_MODE_BTDM);  // ... bluetooth
                        esp_ble_gap_set_scan_params(&ble_scan_params); // reset parameters
			esp_ble_gap_start_scanning(10);
            break;
        default:
            break;
        }
        break;
    }

    default:
        break;
    }
}

void ble_client_appRegister(void)
{
    esp_err_t status;
	printf("Register callback\n");

    // register the scan callback function to the gap module
    if ((status = esp_ble_gap_register_callback(esp_gap_cb)) != ESP_OK) {
        ESP_LOGE(GATTC_TAG, "gap register error, error code = %x", status);
        return;
    }
	
	esp_ble_gap_set_scan_params(&ble_scan_params); // set params
}

void gattc_client_test(void)
{
    esp_bluedroid_init();
    esp_bluedroid_enable();
    ble_client_appRegister();
}

void app_main()
{
	// Initialize NVS.
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);
    esp_bt_controller_enable(ESP_BT_MODE_BTDM);

    gattc_client_test(); 
}
```

hiik3n, 2017-08-24:
@esantosjr great! Reset bt_controller each time helps to increase the performance of scanning surround ble.
Thank you for your suggestion.
However, the number of scanned beacon is not stable. I have 10 beacons advertising every 350ms. I set duration = 10s, internal = 5s and window = 2.5s. The result of each scan is not stable,  maybe 10 or 7 or 2 or 1 or even 0 for some perioid of time (some minutes then normally)

How your scan result performance?

esantosjr, 2017-08-30:
@hiik3n the performance is good, I can scan the devices many times during one period.

hiik3n, 2017-08-31:
@esantosjr you can see my esp scanning performance, it look not ok 
 
[image]

hiik3n, 2017-09-08:
@esantosjr @TianHao-Espressif I just found that if I run ble scan only (wifi is not active), I can scan ble for long time. 
But if I active wifi and send scanned ble via mqtt (https://github.com/eclipse/paho.mqtt.embedded-c ), the issue of scan 0 ble package happen. I am trying to find the reason and the solution for that. I also think if the reason is conflict of bluetooth and wifi task, because they use same antenna.

esantosjr, 2017-09-08:
@hiik3n I am facing problems using this library for mqtt: https://github.com/knolleary/pubsubclient

But the issue of scan 0 ble package isn't happen, my problem is more in relation to the connection between the board and wifi. After some time, the connection is lost and when there is a reconnection the board halt. There is no error message, and nothing is printed.

FayeY, 2017-09-26:
Hi, have you tried the latest IDF? Is this problem still unsolved?

TianHao-Espressif, 2017-09-26:
reference to #924,  asker have tested the latest ESP-IDF. It can work.

closed 2017-11-21 by FayeY

burkulesomesh43, 2018-08-08:
@esantosjr 
I change my code as you given.  but gives an error..  

error: too many arguments to function 'esp_bt_controller_disable'
                              esp_bt_controller_disable(ESP_BT_MODE_BTDM); // reinit ..

esantosjr, 2018-08-08:
@burkulesomesh43, maybe you are using a newer version of the library and that function has changed.

You can check the function's definition and see which are the parameters that it expects.
Also, the issue #924 should be helpfull, as said in the last commentary.

burkulesomesh43, 2018-08-10:
@esantosjr, yes I am using latest version.
I got solution. actually esp_bt_controller_disable() function in my liabrary returns void. so no need to give ESP_BT_MODE_BTDM parameter to function. the function itself include esp_bt_get_mode() function.
now it's working properly.
thank you for your quick response.

esantosjr, 2018-08-10:
@burkulesomesh43 nice! I'm glad I could help.