espressif/esp-idf #912: APSTA other device connet AP,can't access network

zhao007z8, 2017-08-16:
esp32 device open APSTA.  and can access network. but when phone connect the device AP. can't access network.  how to config? have called  esp_wifi_connect  in SYSTEM_EVENT_STA_START
void wifi_init_test(char *ssid,char *password)
{
	wifi_config_t sta_config = {0};
	strcpy((char *)sta_config.sta.ssid,ssid);
	strcpy((char *)sta_config.sta.password,password);

	wifi_config_t ap_config = {
        .ap = {
            .ssid = "test888",
            .password = "88888888",
            .ssid_len = 0,
            .max_connection = 4,
            .authmode = WIFI_AUTH_WPA_PSK
        }
    };

    nvs_flash_init();	
    tcpip_adapter_init();
    ESP_ERROR_CHECK( esp_event_loop_init(wifi_event_handler, NULL) );

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
   ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
   ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_APSTA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_config) );
    esp_err_t tmp =  esp_wifi_set_config(ESP_IF_WIFI_AP, &ap_config);
    ESP_ERROR_CHECK( esp_wifi_start() );
}

liuzfesp, 2017-08-21:
Hi @zhao007z8 what do you mean "can't access network"? Do you mean can't get IP address, or can't access internet?

FayeY, 2017-09-18:
Hi @zhao007z8 , please have a try with the latest IDF, this issue should have been solved in the latest IDF. Thanks for your interest in ESP32!

closed 2017-11-21 by FayeY