espressif/esp-idf #913: ESP32 Wi-Fi Support for 802.11ah HaLow

GarageGadget, 2017-08-16:
Hi,

Will the Espressif ESP32 support the Wi-Fi 802.11ah (900MHz) HaLow standard in the future?

http://www.wi-fi.org/news-events/newsroom/wi-fi-alliance-introduces-low-power-long-range-wi-fi-halow

BR,

negativekelvin, 2017-08-16:
I don't think so, it doesn't have a 900mhz radio

closed 2017-08-16 by GarageGadget

CarlosGS, 2017-08-16:
I'm sure @cnlohr would be able to implement this by bit-banging a paper clip

cnlohr, 2017-09-09:
https://pics.me.me/your-scientists-were-so-preoccupied-with-whether-they-could-that-19308558.png

Thunderbolt2-o, 2022-01-21:
I just now came across this technology is as this 2022  is there any chance of getting HaLow support to esp32

MaxwellAlan, 2022-03-04:
@Thunderbolt2-o @GarageGadget We are also following the development of 802.11ah HaLow, but still not in our schedule now.

5t0n3r1pp3r, 2023-12-28:
is There any Hardware by espressif that supports it?

MaxwellAlan, 2023-12-28:
Hi @5t0n3r1pp3r 

> is There any Hardware by espressif that supports it?

Nope :)

MaxwellAlan, 2024-07-16:
Hi @beyonlo 

Thank you for your continued attention, still not in our plan :)

TiDiChi, 2024-07-31:
> > Hi @beyonlo
> > Thank you for your continued attention, still not in our plan :)
> 
> Hello @MaxwellAlan
> 
> Could you please explain a bit why espressif do not have plans for that? Maybe we (users/developers) are thinking that Halow is something amazing, that will solve many problems, but maybe technically Halow is not so good on espressif vision.
> 
> Example: Halow is subGz (~900MHz), it means that signal can pass through walls and so on, where 2.4GHz and 5GHZ can't. There are also big advantages for low power, few data, very longer distances, etc, but maybe the Espressif maybe see some disadvantages, like as need a special AP (with subGHz), and Espressif can consider for low power and a few longer distances already exists ESP-NOW (even in 2.4/5Ghz), or Espressif can see that will not possible/or difficult to build a SoC with microcontroller + Halow like as is done with WiFi 4/6.
> 
> Anyway, I see that very few brands are developing a SoC using Halow, and it may be the Espressif and all those brands are thinking the same thing - what are those thoughts?
> 
> Thank you in advance!

I have the same questions and I am looking forward to a response from espressif

TiDiChi, 2024-07-31:
@MaxwellAlan I have high hopes for this wireless technology. If you (or espressif) are having a plan which develop a soc with wifi-halow and need help, i am willing to contribute to that project.

cnlohr, 2024-07-31:
There are a great multitude of wireless protocols with benefits and drawbacks of each.  If espressif wanted to open the floodgates, they'd just let their hardware be used in a more configurable way, but that's not their jam.  They make their own magic sauce and make it really magic... and let other companies find their stride.  Let 'em be.

There's a lot of technical reasons why it doesn't match their jam, ranging from difficulty making a completely integrated (no need for any external components) stuff at 900 MHz... sure, espressif might find some secret sauce here whooo knowwws, but if they aren't inspired to make a real gem, I don't want to the part see the market.

Besides, there are lots of other protocols around 900MHz geared for long range. Most geared for lower bandwidth, and I don't think anyone wants broad spectrum 900MHz iot stuff gumming it up.