espressif/esp-idf #914: FreeRTOS call vTaskDelete crashes after multiple task creation and deletion

vwitti, 2017-08-16:
I started a FreeRTOS on any core and deleted this task. If I repeat the creation and deletion of this task without pause, FreeRTOS crashes. If I add a pause of at least 3 ticks after vTaskDelete than the code works. This seems to be a bug inside the ESP32 FreeRTOS implementation. Please see attached example. I tested this problem, with today master of the esp-idf.
This seems to be a timing issue inside the freertos implementation

[image]

projectgus, 2017-08-17:
Hi @vwitti ,

To save us time reproducing this, could you please post the code snippet somewhere as text? https://gist.github.com is one good place.

projectgus, 2017-08-17:
From just looking at the code, the problem (I think) is to do with reusing the same static TCB for the second task, before the first task has been cleaned up. Task cleanup in FreeRTOS is performed by the idle task, so the idle task has to run before the memory is actually freed.

This is a behaviour of the FreeRTOS architecture & design (and documented for `vTaskDelete()`), and I think it will be hard to change. Some potential workarounds:

* Allocate these tasks dynamically rather than statically.
* Use different buffers for the two TCBs (you should be able to safely reuse the task stack, because the task will not be scheduled again after vTaskDelete() is called.)

Otherwise, sleeping until the idle task has run while allow you to reuse the TCB after this happens. However, knowing for sure when the idle task has run can be difficult. Probably the best method I can think of is:
```
vTaskPrioritySet(NULL, tskIDLE_PRIORITY);
vTaskDelay(2);
vTaskPrioritySet(NULL, old_priority);
```

Or you can use `vTaskSetThreadLocalStoragePointerAndDelCallback()` to set a dummy thread-local pointer and a callback, which will run (in the context of the idle task) when the task is finally deleted.

projectgus, 2017-08-17:
One more possibility, if you need to recreate a task reusing the same stack & TCB then perhaps refactor the code so you only ever have a single task, and use a queue or another RTOS data structure to push different "main" routines or jobs into it.

igrr, 2017-09-07:
Hi @vwitti, do you have any progress on this issue?

closed 2017-11-21 by FayeY