espressif/esp-idf #916: OTA Hangs ESP32

pjjasinski, 2017-08-18:
Hello
My project hangs on operation:
err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);

partitions.csv
# Espressif ESP32 Partition Table
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
factory,  0,    0,       0x10000, 512K
ota_0,    0,    ota_0,  , 512K
ota_1,    0,    ota_1,  ,512K

FayeY, 2017-08-21:
Hi, here is an example of [OTA](https://github.com/espressif/esp-idf/tree/master/examples/system/ota), have you tried it?

pjjasinski, 2017-08-21:
Yes i have tried it and its working ok

pjjasinski, 2017-08-21:
When i change OTA example in menuconfig to use partitions.csv
and then make partition_table, make erase_flash, make flash monitor
I get this:

I (3752) ota: Server IP: 192.168.2.250 Server Port:8070
I (3762) ota: Connected to server
I (3762) ota: Connected to http server
I (3762) ota: Send GET request to server succeeded
I (3772) ota: Writing to partition subtype 16 at offset 0x90000
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40116f74  PS      : 0x00060d30  A0      : 0x800ed66e  A1      : 0x3ffbe230  
0x40116f74: cnx_rc_update_age at ??:?

A2      : 0x3ffb8864  A3      : 0x3ffbe2c0  A4      : 0x3ffb7338  A5      : 0x3ffb4698  
A6      : 0x00000001  A7      : 0x3ffb8924  A8      : 0x800ecbd5  A9      : 0x3ffbe210  
A10     : 0x3ffb4698  A11     : 0x00000005  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40001699  LEND    : 0x400016aa  LCOUNT  : 0xfffffffa  

Backtrace: 0x40116f74:0x3ffbe230 0x400ed66b:0x3ffbe2b0 0x400ede35:0x3ffbe380 0x400f5931:0x3ffbe3a0 0x400f70ff:0x3ffbe3f0
0x40116f74: cnx_rc_update_age at ??:?

0x400ed66b: sta_input at ??:?

0x400ede35: sta_rx_cb at ??:?

0x400f5931: ppRxPkt at ??:?

0x400f70ff: ppTask at ??:?

Rebooting...

pjjasinski, 2017-08-21:
Ok now OTA example is working with partitions.csv:

>
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory, 0x10000,  1M
ota_0,    app,  ota_0,   ,         1M
ota_1,    app,  ota_1,   ,         1M

pjjasinski, 2017-08-21:
Ok looks like i have too small size of ota partition.
For now problem is solved

closed 2017-08-21 by pjjasinski