espressif/esp-idf #917: UART 1 ms timeout following empty TX FIFO

alien-explosive, 2017-08-18:
We are attempting to send a stream of UART data with short timeouts less than 200 uS.

We find that whenever the TX FIFO is empty, the next data byte is not transmitted until approximately 1 millisecond after loading the next character.

We expected that the UART should transmit immediately, or within a few uS.

Our code runs in an interrupt. Snippet below:

```c
void IRAM_ATTR timer_isr(void* arg)
{
	TIMER_ENTER_CRITICAL(&timer_spinlock[TIMER_GROUP_0]); //Pause timer
	TIMERG0.hw_timer[TIMER_0].config.enable = 0;
	TIMER_EXIT_CRITICAL(&timer_spinlock[TIMER_GROUP_0]);
    TIMERG0.int_clr_timers.t0 = 1;
    TIMERG0.hw_timer[0].config.alarm_en = 1;
    if(send_break){
    		GPIO.out1_w1ts.data = 2; //Turn on the LED pin
        TIMER_ENTER_CRITICAL(&timer_spinlock[TIMER_GROUP_0]); //Reset timer alarm value
        TIMERG0.hw_timer[TIMER_0].alarm_high = (uint32_t) (TIMER_DATA_TICKS >> 32);
        TIMERG0.hw_timer[TIMER_0].alarm_low = (uint32_t) TIMER_DATA_TICKS;
        TIMER_EXIT_CRITICAL(&timer_spinlock[TIMER_GROUP_0]);
				send_break = false;
    }
    else{
    		GPIO.out1_w1tc.data = 2; //Turn off LED pin
        TIMER_ENTER_CRITICAL(&timer_spinlock[TIMER_GROUP_0]); //Reset timer alarm value
        TIMERG0.hw_timer[TIMER_0].alarm_high = (uint32_t) (TIMER_BREAK_TICKS >> 32);
        TIMERG0.hw_timer[TIMER_0].alarm_low = (uint32_t) TIMER_BREAK_TICKS;
        TIMER_EXIT_CRITICAL(&timer_spinlock[TIMER_GROUP_0]);

        for (int i = 0; i < 5; i++) { //Loop through the data and write it directly to the register
				WRITE_PERI_REG(UART_FIFO_AHB_REG(uart_num), data[i]);
		}
		for (int i = 0; i < 5; i++) {
			WRITE_PERI_REG(UART_FIFO_AHB_REG(uart_num), data[i]);
		}

    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[10], PIN_FUNC_GPIO);
		send_break = true;
    }

  timer_num++;
	TIMER_ENTER_CRITICAL(&timer_spinlock[TIMER_GROUP_0]); //Re-start timer
	TIMERG0.hw_timer[TIMER_0].config.enable = 1;
	TIMER_EXIT_CRITICAL(&timer_spinlock[TIMER_GROUP_0]);
}
```

[image]

alien-explosive, 2017-08-23:
We found a solution. The problem is that by default an additional TX idle time is enforced in hardware. The following line of code sets the idle time to zero:

`UART[uart_num]->idle_conf.tx_idle_num = 0;`

benpeoples, 2017-08-23:
This is the best news all day!

closed 2017-10-31 by FayeY