espressif/esp-idf #918: No make/project.mk in the toolchain

ayhoung, 2017-08-19:
Hi all,

I followed the readme documentation on getting started to the exact letter (I think) but I still seem to be missing something. whenever I run the make command in the hello_world example, I get this error:
```
Makefile:8: ~/esp/xtensa-esp32-elf/bin/make/project.mk: No such file or directory
```

my path is set up correctly but this is all I see under `~/esp/xtensa-esp32-elf/bin/`:
```
drwxr-xr-x@ 31 allen  staff     1054 Jan 11  2017 .
drwxr-xr-x@ 10 allen  staff      340 Aug 19 11:41 ..
-rwxr-xr-x@  1 allen  staff   865892 Jan 11  2017 xtensa-esp32-elf-addr2line
-rwxr-xr-x@  1 allen  staff   892476 Jan 11  2017 xtensa-esp32-elf-ar
-rwxr-xr-x@  1 allen  staff  1203416 Jan 11  2017 xtensa-esp32-elf-as
-rwxr-xr-x@  1 allen  staff   817776 Jan 11  2017 xtensa-esp32-elf-c++
-rwxr-xr-x@  1 allen  staff   861688 Jan 11  2017 xtensa-esp32-elf-c++filt
lrwxr-xr-x@  1 allen  staff       20 Jan 11  2017 xtensa-esp32-elf-cc -> xtensa-esp32-elf-gcc
-rwxr-xr-x@  1 allen  staff   817776 Jan 11  2017 xtensa-esp32-elf-cpp
-rwxr-xr-x@  1 allen  staff     2910 Jan 11  2017 xtensa-esp32-elf-ct-ng.config
-rwxr-xr-x@  1 allen  staff    31976 Jan 11  2017 xtensa-esp32-elf-elfedit
-rwxr-xr-x@  1 allen  staff   817776 Jan 11  2017 xtensa-esp32-elf-g++
-rwxr-xr-x@  1 allen  staff   817776 Jan 11  2017 xtensa-esp32-elf-gcc
-rwxr-xr-x@  1 allen  staff   817776 Jan 11  2017 xtensa-esp32-elf-gcc-5.2.0
-rwxr-xr-x@  1 allen  staff    28340 Jan 11  2017 xtensa-esp32-elf-gcc-ar
-rwxr-xr-x@  1 allen  staff    28340 Jan 11  2017 xtensa-esp32-elf-gcc-nm
-rwxr-xr-x@  1 allen  staff    28340 Jan 11  2017 xtensa-esp32-elf-gcc-ranlib
-rwxr-xr-x@  1 allen  staff   442828 Jan 11  2017 xtensa-esp32-elf-gcov
-rwxr-xr-x@  1 allen  staff   389912 Jan 11  2017 xtensa-esp32-elf-gcov-tool
-rwxr-xr-x@  1 allen  staff  4808944 Jan 11  2017 xtensa-esp32-elf-gdb
-rwxr-xr-x@  1 allen  staff   933400 Jan 11  2017 xtensa-esp32-elf-gprof
-rwxr-xr-x@  1 allen  staff  1205000 Jan 11  2017 xtensa-esp32-elf-ld
-rwxr-xr-x@  1 allen  staff  1205000 Jan 11  2017 xtensa-esp32-elf-ld.bfd
-rwxr-xr-x@  1 allen  staff   878208 Jan 11  2017 xtensa-esp32-elf-nm
-rwxr-xr-x@  1 allen  staff  1073208 Jan 11  2017 xtensa-esp32-elf-objcopy
-rwxr-xr-x@  1 allen  staff  1197724 Jan 11  2017 xtensa-esp32-elf-objdump
-rwxr-xr-x@  1 allen  staff   892500 Jan 11  2017 xtensa-esp32-elf-ranlib
-rwxr-xr-x@  1 allen  staff   443744 Jan 11  2017 xtensa-esp32-elf-readelf
-rwxr-xr-x@  1 allen  staff   865944 Jan 11  2017 xtensa-esp32-elf-size
-rwxr-xr-x@  1 allen  staff   865948 Jan 11  2017 xtensa-esp32-elf-strings
-rwxr-xr-x@  1 allen  staff  1073208 Jan 11  2017 xtensa-esp32-elf-strip
```

negativekelvin, 2017-08-19:
You add the toolchain binary to your PATH but what is your IDF_PATH

http://esp-idf.readthedocs.io/en/latest/get-started/add-idf_path-to-profile.html

ayhoung, 2017-08-19:
Thanks negativekelvin!

closed 2017-08-19 by ayhoung