espressif/esp-idf #919: Ping library for esp idf

Nakul93, 2017-08-19:
Does anyone have a custom ping library for wifi in esp32?

projectgus, 2017-08-20:
Hi @Nakul93,

There is ICMP Ping support built into IDF, you can include esp_ping.h and use the functionality provided there:
https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/ping/esp_ping.h

For questions like this about IDF, the best place to ask is the forum http://esp32.com . If you have follow-up questions about ping functionality, please ask there. The Issues list is for reporting bugs and making specific feature requests for IDF itself.

Angus

closed 2017-08-20 by projectgus

MightyPork, 2018-10-24:
do we have any example of how to use this API? I didn't find anything in examples or elsewhere online (in fact I implemented it myself, before stumbling upon this file in idf). thanks