espressif/esp-idf #920: Incorrect Assertion in returnItemToRingbufDefault

HendrikSP, 2017-08-19:
**Problem:**

The assertion in returnItemToRingbufDefault (line 340)
```
configASSERT(data < rb->data+rb->size);
```
is wrong and needs to be
```
configASSERT(data <= rb->data+rb->size);
```

**Explanation:**

In `getItemFromRingbufDefault` the pointer is set to
```
ret=rb->read_ptr+sizeof(buf_entry_hdr_t)
```
If `hdr->len` is 0 then `rb->read_ptr` could be `rb->data+rb->size-sizeof(buf_entry_hdr_t)` and
the assertion  above will fail on returning the item.

If fixed the assertion locally and everything runs fine.

Spritetm, 2017-08-28:
Gotcha. Interesting edge case, this. The fix is in our internal queue.

igrr, 2017-09-07:
Fixed in 4da7ee1295172fb0f64d8cfa1711170ed782ca42.

closed 2017-09-07 by igrr