espressif/esp-idf #922: Feature request: I2C clear bus function in driver.

PerMalmberg, 2017-08-20:
Quote from the [I2C-bus specification and user manual, page 20, §3.1.16](http://www.nxp.com/docs/en/user-guide/UM10204.pdf)

> If the data line (SDA) is stuck LOW, the master should send nine clock pulses. The device
that held the bus LOW should release it sometime within those nine clocks.

Would it be possible to have this implemented in the driver? It'd be a much more elegant solution than having to de-initialize, setup GPIO, generate clock pulses, the reinitialize the I2C driver.

I imagine having a signature as follows, so that it is possible to clear each bus separately: 
```C++
bool i2c_clear_bus(i2c_port_t port_num, i2c_cmd_handle_t cmd_handle)
```

CarlosGS, 2017-08-21:
Related to https://github.com/espressif/arduino-esp32/issues/349

costaud, 2017-10-18:
If, only the data wire is held by the slave, the master can still send some data to the bus(although the sda won't change), so that there will be clock on scl wire to release the slave?  Is it correct?

costaud, 2017-10-19:
OK, according to my test result, if the SDA is stuck, master can not send clock out, but the code can be simple like this to switch signal, generate clock on SCL, then set back the signals. 

```
    gpio_set_direction(19, GPIO_MODE_OUTPUT); //scl
    gpio_set_direction(18, GPIO_MODE_OUTPUT); //sda
    for (int i = 0;i < 9; i++) {
        gpio_set_level(19, 1);
        gpio_set_level(19, 0);
    }
    i2c_set_pin(i2c_num, 18, 19, 1, 1, I2C_MODE_MASTER);
```

PerMalmberg, 2017-10-23:
Oh, I see we now have a i2c_master_clear_bus(). Awesome, @igrr  :)

closed 2017-10-23 by FayeY