espressif/esp-idf #924: Bluetooth Controller: LE Scanning Stops after 1-2 hours with BTstack

mringwal, 2017-08-21:
When running the simple "gap_le_advertisement" demo from BTstack with some LE devices in range, the reported advertisements stop after 1-2 hrs. There's no crash on the console or other unexpected HCI events. A BTstack heartbeat timer still 'beats' every few seconds. 

Could someone at Espressif run a similar test with Bluedroid or the BTstack gap_le_advertisements example? The code for processing a single advertisement in BTstack is pretty simple and fully deterministic, so I suspect an issue in the Bluetooth Controller implementation.

It was reported to the BTstack issue tracker (https://github.com/bluekitchen/btstack/issues/60) and I was able to reproduce it.

hiik3n, 2017-08-31:
@mringwal how is the issue now?
I also tried the gap_le_advertisements example. first time it stopped after 2hours, 2nd time it stopped after 6hours.
I am thiniking about scaning in some period of time (may be 5s then stop for 5s). I also think about reset the configuration for the btstack for each duration scan (but not sure how to proccess...)
you have any idea?

mringwal, 2017-08-31:
@hiik3n did you see this issue also with the Bluedroid stack?

The easiest hack to work around this probably would be to reboot every hour.

hiik3n, 2017-09-08:
@mringwal I also tried bluedroid, I experienced that bluedroid can scan ble better than btstack (in case of number of scanned ble package and how long it "live" )
with bluedroid, I can scan ble for days (however, when I combine bluedroid with wifi + mqtt, sometime esp32 can scan some ble, sometime not... I am finding solution for synchronize bluedroid + wifi task)
with btstack, I found that number of scanned ble significant decrease (lost a lot of data) so I used bluedroid instead.

TianHao-Espressif, 2017-09-25:
Hi, @mringwal , @hiik3n , there's real a bug of that ESP32 cannot receive advertising after a long time scan. But it's fixed in latest master. I think that it may be related to your problem.

mringwal, 2017-09-25:
@TianHao-Espressif sounds good. I did update and start a scan this morning, and it's still working (it's past tea-time now). Thanks for fixing this issue.

closed 2017-09-25 by mringwal

beats0126, 2018-05-16:
@TianHao-Espressif may i know how to get the latest master? is it the file is under this link [https://github.com/espressif/esp-idf](url) ? Now my working stopped because meeted this scan 0 result. it happen after i used esp.restart() or press on the reset button.

beats0126, 2018-05-16:
@TianHao-Espressif  i get the file already. thanks, i will try up it with my current build.

beats0126, 2018-05-16:
@TianHao-Espressif  may i know why after i comiling file, i using the esp32 dev module. when the message showing
Uploading esp32Sketch to "ESP32 Dev Module" using "Com3"
The system cannot find the file specified
do u know what causing?

beats0126, 2018-05-17:
@TianHao-Espressif  thanks for your replying.
i am solved my problem just now. now already 2 hour dont happen scan 0 result problem.
my solution is i tried copy the latest core, library and tools/sdk file to my current espressif/esp32 file then the problem
Uploading esp32Sketch to "ESP32 Dev Module" using "Com3"
The system cannot find the file specified
do no happen again to me.
the latest library maybe solved my scan result problem. i still testing on it. thanks again!