espressif/esp-idf #925: btc_dm_load_ble_local_keys used even if SMP disabled

lucazader, 2017-08-21:
Hi all,

In the latest master fc85cb6 there is a compilation error if Include BLE security module(SMP) is disabled in the kconfig.

`In btc_main.c`
```
static void btc_init_bluetooth(void)
{
    osi_alarm_create_mux();
    osi_alarm_init();
    bte_main_boot_entry(btc_init_callback);
    btc_config_init();
    //load the ble local key whitch has been store in the flash
    btc_dm_load_ble_local_keys();
}
```
`btc_dm_load_ble_local_keys` is used without an `#if (SMP_INCLUDED == TRUE)` guard. This will cause a compilation error if SMP is disabled in the kconfig, as the function declaration in the file `btc_ble_storage.h` is excluded from the build using `#if (SMP_INCLUDED == TRUE)`

lucazader, 2017-08-21:
This is also true for the following functions:

`btc_ble_clear_bond_device`
`btc_ble_get_bond_device_list`
`btc_ble_remove_bond_device`

Yulong-espressif, 2017-08-22:
@lucazader Thanks for your suggestions, I have update the code, please update the SDK and try again, if any question, please let me know. Thank you so so much.

lucazader, 2017-08-22:
Hi @Yulong-espressif 

Thanks for your response!
I look forward to seeing this commit being merged into the github master.

lucazader, 2017-09-20:
Fixed at some point recently (cant seem to find the exact commit)

closed 2017-09-20 by lucazader