espressif/esp-idf #926: changing wifi mode no longer works after adding fast crypto

CristiCimpianu, 2017-08-22:
In my particular scenario I need to change from one mode to another so I'm doing esp_wifi_init once and then if I want to go to station mode:
```
esp_wifi_stop();
esp_wifi_set_mode (WIFI_MODE_STA);
esp_wifi_set_config (ESP_IF_WIFI_STA, wifi_config);
esp_wifi_start ();
```
and if I want to go to AP mode:
```
esp_wifi_stop();
esp_wifi_set_mode (WIFI_MODE_STA);
esp_wifi_set_config (ESP_IF_WIFI_STA, wifi_config);
esp_wifi_start ();
```
After the commit introducing fast crypto my device no longer connects to my AP unless I remove the call to `esp_wifi_stop()`. Of course, if I remove the call to `esp_wifi_stop()` the initial connection works, but changing the mode no longer does.

XinDeng11, 2017-08-30:
Hi Cristi, I try to reproduce the phenmenon but failed, I can connect both in AP and STA mode, ever I switch to another mode, could you tell me more detailly? Like how you switch the mode, and in which type you can not connect.

XinDeng11, 2017-08-30:
@CristiCimpianu

CristiCimpianu, 2017-08-30:
Hi @XinDeng11 and thanks for your reply.
Let's start with something working which is the power_save example under the wifi folder.

One thing that can be done in order to make it stop working is to add an esp_wifi_stop() call before:
```
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
```
It will remain hanging after the following logs:
```
I (404) power_save: esp_wifi_set_ps().
I (404) power_save: SYSTEM_EVENT_STA_START
I (414) wifi: sleep disable
```
This was working up until the fast_crypto changes.
My real scenario is rather complicated:
1. I initialize both wifi and bt.
2. I put the wifi in STA mode and start an AP scan
3. When the AP scan is done I change the default IP of the AP mode and move wifi to AP mode
4. A client connects either via wifi or via BT and sends me a ssid:password pair
5. I stop BT and I switch the wifi to STA mode to connect using the received information.
The final step of moving to STA mode is getting stuck after these logs:
```
I (52792) wifi: station: 10:68:3f:74:03:37 leave, AID = 1
I (52793) wifi: n:6 0, o:6 0, ap:6 1, sta:255 255, prof:6
I (52798) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (52803) wifi: mode : sta (30:ae:a4:03:50:18)
```
Could you please share how you switch from one mode to the other?

XinDeng11, 2017-08-31:
@CristiCimpianu Hi Cristi, my simple test is base on the power saving example too. I only add switch mode in the wifi_power_save() in power save.c, like 
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
wifi_config_t wifi_config = {
     .sta = {
          .ssid = DEFAULT_SSID,
          .password = DEFAULT_PWD
     },
};
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
ESP_ERROR_CHECK(esp_wifi_start());

ESP_LOGI(TAG, "Switch to AP mode");
wifi_init_config_t wifi_ap_config = {
     .ap = {
          .ssid = "espressif",
          ...
          .beacon_interval = 100,
     },
};

ESP_ERROR_CHECK(esp_wifi_stop());
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_ap_config));
ESP_ERROR_CHECK(esp_wifi_start());

vTaskDelay(1000);

ESP_LOGI(TAG, "Switch to STA mode");

ESP_ERROR_CHECK(esp_wifi_stop());
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
ESP_ERROR_CHECK(esp_wifi_start());

In fact, the fast_crypto function is register when esp_wifi_init() invoke, and the esp_wifi_stop() doesn't cancle the fast crypto function, it should not infulence your switch mode operation.

CristiCimpianu, 2017-08-31:
Hi @XinDeng11 ,
I never meant to say that the fast crypto itself is the problem because I tried replacing the functions with the old ones and got the same behavior. I guess something changed in the wifi library in this commit.
Anyway, since your example worked for me also, I realized the only issue seems to be calling esp_wifi_stop() in the initial path, before any call to esp_wifi_start(). This is what I mentioned in my previous comment and since I can check the current mode of the wifi module this issue has suddenly become minor.
Thanks a lot for the help.

closed 2017-09-07 by igrr