espressif/esp-idf #927: [SPI Slave] last couple of bytes not transferring correctly

ShendoXT, 2017-08-22:
I have 2 esp32s (Wroom32 on a devboard) connected to each other via SPI bus. 
I've compiled SPI sender and receiver example and it seems to be working correctly.

The messages that are sent in the example (ASCII strings) are way shorter than the supplied buffer.
If I try to use the whole buffer for transferring data, the last couple of bytes (eg. 5 bytes).
come up garbled.

Just for a test I shorted the messages to 30 bytes and stopped using DMA on receiver side
but it did not help. Same thing happens. Same with other modes and flags.

Issue seems to be on the receiver side because I communicated with another slave device
and it is working properly.

Spritetm, 2017-08-24:
Did you take heed of the following warning in the [documentation](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/spi_slave.html)?

> Warning: Due to a design peculiarity in the ESP32, if the amount of bytes sent by the master or the length of the transmission queues in the slave driver, in bytes, is not both larger than eight and dividable by four, the SPI hardware can fail to write the last one to seven bytes to the receive buffer.

ShendoXT, 2017-08-24:
Hi, thanks for your reply.

Yes, I'm aware of mentioned limitation in the slave driver.
Seems like I made a typo, I meant to say 32 bytes which to my understanding is within specs.

I uploaded modified samples to a [dedicated repository](https://github.com/ShendoXT/esp32-spi-tests).
Both sender and receiver should display the same data but receiver comes short a few bytes.

ESP32DE, 2017-08-25:
@ShendoXT 
Što ESP-IDF inačica uzimate?

ShendoXT, 2017-08-25:
@ESP32DE
I'm using the master branch.

L-eeX, 2017-11-04:
@ShendoXT 
Hello, I recently development projects also need to use spi slave, but I debug for more than half a month, still no progress, the slave receives the data is not very stable, error rate of more than half, every time I send and receive only four bytes, but in this way, I don't know your spi slave can be used? Can you tell me if you have any good advice?Thank you!

Mahmoudomar91, 2018-05-18:
i am having the same problem with the code. 
Did anyone find a solution?? 
This is an image for what comes at the serial monitor
[image]