espressif/esp-idf #930: Major security flaw: Device connects to OPEN network when configured for WPA

dsiganos, 2017-08-23:
To setup an ESP32 module to connect to WPA we set an SSID and a passphrase.
However, the device will also connect to an open network with the same SSID.

This is a major security flaw because someone can introduce an open wifi router as a man-in-the-middle attack and get access to all the packets very easily.

Test case that breaks:
1) Setup an AP with the following details:
SSID = myssid
PASSPHRASE = mypassword
Encryption = WPA2

2) Then configure the ESP32 to connect to network myssid/mypassword and check that it connects.

3) Now change the configuration of the AP to be:
SSID = myssid
Encryption = OPEN/No encryption

4) Check that module still connects. You will find that it does and it shouldn't!

XinDeng11, 2017-09-28:
@dsiganos Hi dsiganos, thank you for you feedback, we have opimized the scan mechanism and could make sure connect to a saver one, it will official release in the v3.0 idf, you can see the scan example which added, thanks.

FayeY, 2017-10-16:
Please have a try with the latest IDF, this issue should have been solved. If not, please feel free to reopen it.

closed 2017-10-16 by FayeY

burkulesomesh43, 2019-08-30:
Hi @FayeY,
I am using esp idf v3.2 still I am getting same issue.

Thanks,
Somesh Burkule

liuzfesp, 2019-08-30:
Hi @burkulesomesh43, could you provide your commit ID please? I think this issue should have been fixed long time ago.

burkulesomesh43, 2019-08-30:
my commit ID: 055943e29346e77c50589c61a8a26101a8b35d7b

burkulesomesh43, 2019-08-30:
Hi @liuzfesp,
commit ID:  055943e29346e77c50589c61a8a26101a8b35d7b

esp-HarveyRong, 2019-11-07:
Hi @burkulesomesh43 Is the problem still there? I use the same commit ID and didn't reproduce the problem.

burkulesomesh43, 2019-11-07:
Hi @esp-HarveyRong,
Yes there is still problem. now I am using esp-idf [v3.3(](https://github.com/espressif/esp-idf/releases/tag/v3.3) ).

burkulesomesh43, 2019-11-07:
Hi @esp-HarveyRong,
It is resolved. I am facing issue with WEP, WPA and WPA2.
esp connects to WPA2-PSK but configured as WEP or WPA.

Alvin1Zhang, 2019-11-08:
@burkulesomesh43 Thanks for the updates. For the new issue, you could file a separate ticket for better tracking. Thanks.

esp-HarveyRong, 2019-11-27:
Hi @burkulesomesh43 Can you describe your issue in detail? It will help me reproduce and debug.E.g. commit ID