espressif/esp-idf #931: Security flaw: ESP32 connects to WEP router

dsiganos, 2017-08-23:
To setup an ESP32 module to connect to WPA we set an SSID and a passphrase.
However, the esp32 will also try to connect to a WEP secured network with the same SSID.

The problem is due to the WiFi being overly simplistic and not being able to describe the type of encryption we want.

This is a security flaw because it is possible to downgrade the security of the link to WEP instead of the WPA that the user probably wanted.

This issue is related to https://github.com/espressif/esp-idf/issues/930

Test case that breaks:
 1)   Setup an AP with the following details:
    SSID = myssid
    PASSPHRASE = ABCDEFABCDEFA
    Encryption = WPA2

 2)   Then configure the ESP32 to connect to network myssid/ABCDEFABCDEFA and check that it connects.

 3)   Now change the configuration of the AP to be:
    SSID = myssid
    WEP KEY = ABCDEFABCDEFA
    Encryption = WEP

4)    Check that the esp32 still connects. You will find that it does and it shouldn't!

ESP32DE, 2017-08-27:
@dsiganos 

you need 2 AP at same time for this test:
-----------------------------------------
what does the esp32 connect if you drive at same time the AP(0) with WPA and AP(1) with WEB,
the AP's have different channel: 
if you setup the ESP32 ( client mode right? ) for a wanted AP  
-  for same channel like the WPA AP
-  for same channel like the WEB AP

is the connect cross too`?

best wishes
rudi ;-)

dsiganos, 2017-08-28:
@ESP32DE 
The test you describe is irrelevant. Even if it prefers the WPA over the WEP, which I do not know if it does but I imagine it does, that doesn't make it any better. What it prefers is not the issue here. The issue is that there is no way to stop an attacker from downgrading the connection to WEP, which we know it to be completely broken and the keys can be easily guessed.

The problem is the WiFi API which is too simplistic. We only give it an SSID and passphrase and we have no way to say what encryption we expect it to use.

projectgus, 2017-08-29:
Hi @dsiganos ,

Thanks for reporting this bug. The API should be updated to provide control over these kind of situations.

> The issue is that there is no way to stop an attacker from downgrading the connection to WEP, which we know it to be completely broken and the keys can be easily guessed.

I'm not certain this is a security bug in the way described. In order to downgrade the connection, an attacker still needs to know the original WPA passphrase. If they know the WPA passphrase, they can just as easily create an AP using WPA and gain access to all WiFi traffic that way (the AP always has access to traffic of connected clients).

This is essentially a shortcoming in the WPA-PSK auth technique (regardless of platform). If this threat is a  concern then there are two possible solutions:

* Move to WPA2-EAP (where the AP presents a certificate and can therefore be authenticated by the client).
* Only use secure authenticated secure transport layer protocols (TLS, etc.), in which case the security of the WiFi layer is no longer critical.

However, in any case, this behaviour should be fixed in IDF to prevent unwanted association to WEP APs. Thanks again for reporting it.

dsiganos, 2017-08-29:
>>  The issue is that there is no way to stop an attacker from downgrading the connection to WEP, which we know it to be completely broken and the keys can be easily guessed.

> I'm not certain this is a security bug in the way described. In order to downgrade the connection, an attacker still needs to know the original WPA passphrase. If they know the WPA passphrase, they [...]

I am not certain that the keys can be guessed either. And I am not certain what happens when the WPA passphrase length does not match a WEP key length e.g. 13 bytes. I am assuming that once the connection is WEP,and if the WPA passphrase length is a valid WEP key length, it will be possible to be cracked but I don't really know for sure because it is not the normal WEP attack scenario where the AP is hacked instead of the Station.

I am certain that a connection that is meant to be WPA can be downgraded to a WEP, which is flawed in many respects. And that by itself is a major security concern which you have agreed with, so that's good.

XinDeng11, 2017-10-12:
Hi, @dsiganos we have improved it, the commit ID is d34c5955, all the versions after this should be OK, if there is any queries, please let me know, thanks.

closed 2017-10-16 by FayeY