espressif/esp-idf #933: Running out of memory after updating

seopyoon, 2017-08-24:
Hello,

My code utilises AWS IoT and BLE at the same time, thus having a huge memory usage.
It so afar worked well, but since I updated ESP-IDF lately, I am getting the following error.  

```
E (2838) esp_image: Image length 1092928 doesn't fit in partition length 1048576
E (2891) boot: Factory app partition is not bootable
E (2943) boot: No bootable app partitions in the partition table
```
Could the size of the build have increased after I updated ESP-IDF?  My code is the same, and I have suddenly got the above message, and thus uable to run the exact same code any longer.

Please help me in finding solution to the issue.  Thank you very much.

igrr, 2017-08-24:
You can increase app partition size if you define a custom partition table:
http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html#creating-custom-tables

You just need to add a custom partition table .csv file to your project, and run `make menuconfig` to set the name of this file.

seopyoon, 2017-08-25:
@igrr I added partition table, so it has enough app space (increase from 1M to 1.5M).  But I am experiencing a new problem now.  
My app initialises WiFi stack AND BLE stack, and when received SSID/PASSPHRASE via BLE, it tries to connect to AP.  But after updating (the esp-idf and the partition table), it behaves awkwardly.

Previously, this was the log info I used to get:
```
I (687) WiFi: Connecting to SSID: <SSID_FROM_BLE> // PASS: <PASS_FROM_BLE>
I (687) wifi: wifi timer task: 3ffdd06c, prio:22, stack:3584
I (717) phy: phy_version: 357.0, a6bf95b, Jul 25 2017, 12:28:06, 0, 0
I (717) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (717) wifi: mode : sta (24:0a:c4:05:73:9c)
I (717) AWS: AWS IoT SDK Version 2.1.1-
I (1927) wifi: n:10 2, o:1 0, ap:255 255, sta:10 2, prof:1
I (2577) wifi: state: init -> auth (b0)
I (2587) wifi: state: auth -> assoc (0)
I (2587) wifi: state: assoc -> run (10)
I (2707) wifi: connected with <SSID_FROM_BLE>, channel 10
```

What I get now is,
```
I (122787) WiFi: Connecting to SSID: <SSID_FROM_BLE> // PASS: <PASS_FROM_BLE>
I (122787) wifi: wifi timer task: 3ffd1694, prio:22, stack:3584
I (122797) wifi: Init Ampdu: 1 tx baw=6 rx baw=6
I (122797) wifi: mode : sta (24:0a:c4:05:73:9c)
```
and it **hangs** there.  The difference I can observe is the third line from the log when it was working (the `phy` part).
Could you suggest any idea to fixing the issue, or what may be the problem here.

Thanks in advance.

igrr, 2017-08-25:
@liuzfesp please advise

seopyoon, 2017-08-28:
to provide more information, 

My application flow that **WORKED** before the esp-idf update: 
a) if no wifi information available, enable ble, 
b) receive ssid and passphrase via ble, 
c) try connecting,
d) when 'got ip', save wifi info and restart,
e) because wifi info available, don't start ble yet, establish connection with AWS IoT only (because mbedtls uses too much stack, and concurrent ble & mbedtls didn't work, due to stack overflow)
f) after aws iot connection is established, enable ble stack.

Now, when after receiving ssid/pass from ble, and try to connect to wifi.  no `phy` tagged debug log, as described above.

What I have found is that, the below flow works in the updated esp-idf:
a) if no wifi information available, enable ble, 
b) receive ssid and passphrase via ble, 
**c-1) save wifi info and restart**
e) because wifi info available, don't start ble yet, establish connection with AWS IoT only (because mbedtls uses too much stack, and concurrent ble & mbedtls didn't work, due to stack overflow)
f) after aws iot connection is established, enable ble stack.

This is not a desired flow because users may mistype password or select wrong wifi, hence leading to failed wifi connection.  I wanted to save wifi info only if `GOT IP` is received.  But the above flow leads to successful wifi connection.

So, the difference is that the **undesired, working** flow (one below) tries wifi connection when ble is **not** enabled, and the **used-to-work** flow tries wifi connection concurrently with ble for a short period of time.

Please advice.  Thanks a lot in advance.

seopyoon, 2017-08-28:
Plus, my partition table:

```
# Espressif ESP32 Partition Table
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x6000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory, 0x10000, 0x110000
```

seopyoon, 2017-08-28:
Solved partially:

I had, `ESP_ERROR_CHECK( esp_wifi_stop() );` before starting wifi with the desired SSID/PASS.  By removing that line, it works.  (Again, it used to work with that line included before updating).  

But the problem is, if already started, and if want to connect to a different AP, `esp_wifi_stop()` is necessary.
Is there a way to see if esp_wifi has already started?  If such flag is available, I could call `esp_wifi_stop()` only when flagged.

seopyoon, 2017-08-31:
I put a flag myself, which sets to high on `case SYSTEM_EVENT_STA_START` and sets low on `case SYSTEM_EVENT_STA_STOP` inside the `event_handler`.  If `wifi_started` (the boolean I created) is high, I call `ESP_ERROR_CHECK( esp_wifi_stop() );` before starting wifi with a new ssid and passpharse.

closed 2017-08-31 by seopyoon