espressif/esp-idf #934: new console resets micro in task

paulbearne, 2017-08-24:
I can run the new system console example but put the code inside a freertos task and the module keeps resetting when it registers a command using esp_console_cmd_register the command. don't register a command and the command line works fine prompt stores history etc. iv'e tried to narrow it down with debug messages but seems to fail at differant places in the esp_console_cmd_register i suspect its to do with the memory allocation but having tried most things now at a bit of a loss

igrr, 2017-08-25:
Hi, we have the fix for this issue in the merge queue. In the meantime, you may apply the following patch:

```patch
From 0ed64da963f10fa18a9f0e4bde812930bcdb364d Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Wed, 23 Aug 2017 01:07:03 +0800
Subject: [PATCH] console: initialize buf_size before calling open_memstream

POSIX open_memstream documentation [1] does not mention that it may use
the  value supplied in buf_size before the call.
newlib implementation of open_memstream does use it as a hint of the
buffer size [2]. To avoid using potential garbage in this variable,
newlib caps the size to 64kB (!).

If the allocation of this initial buffer fails, NULL file pointer is
returned. Previous code did not check returned file pointer and crashed
when it was used.

Initialize size to zero (in which case newlib allocates a 64 byte
buffer), and check the returned file pointer.

[1] http://pubs.opengroup.org/onlinepubs/9699919799/functions/open_memstream.html
[2] https://github.com/espressif/newlib-esp32/blob/23c0f21/newlib/libc/stdio/open_memstream.c#L26-L29
    https://github.com/espressif/newlib-esp32/blob/23c0f21/newlib/libc/stdio/open_memstream.c#L324-L336
---
 components/console/commands.c | 10 ++++++----
 1 file changed, 6 insertions(+), 4 deletions(-)

diff --git a/components/console/commands.c b/components/console/commands.c
index a451584d..dfcbf450 100644
--- a/components/console/commands.c
+++ b/components/console/commands.c
@@ -105,11 +105,13 @@ esp_err_t esp_console_cmd_register(const esp_console_cmd_t *cmd)
         asprintf(&item->hint, " %s", cmd->hint);
     } else if (cmd->argtable) {
         /* Generate hint based on cmd->argtable */
-        char* buf;
-        size_t buf_size;
+        char* buf = NULL;
+        size_t buf_size = 0;
         FILE* f = open_memstream(&buf, &buf_size);
-        arg_print_syntax(f, cmd->argtable, NULL);
-        fclose(f);
+        if (f != NULL) {
+            arg_print_syntax(f, cmd->argtable, NULL);
+            fclose(f);
+        }
         item->hint = buf;
     }
     item->argtable = cmd->argtable;
-- 
2.11.0 (Apple Git-81)

```

paulbearne, 2017-08-25:
Thank you patched and working

igrr, 2017-08-28:
Fixed in e0e1c07.

closed 2017-08-28 by igrr