espressif/esp-idf #935: Bluetooth Controller: ACL-LE Packets larger than 27 bytes are sent in 26 byte fragments

mringwal, 2017-08-25:
Without Data Length Extension, ACL-LE packets can have up to 27 bytes PDU over-the-air.

When sending 27 byte ACL-LE packets to the Bluetooth Controller, they are sent as a single 27 byte PDU over the air (as expected). See esp32-streamer-csr-client-27-byte-pdu.psd air trace from TI's Bluetooth Sniffer (http://www.ti.com/tool/packet-sniffer).

When sending larger (e.g. 200 byte) ACL-LEs, the Bluetooth Controller must sent them in 27 byte fragments. However, it only sends 26 byte ACL-LE packets over the air. See esp32-streamer-csr-client-26-byte-pdu.psd in the .zip.

Sending 26 instead of 27 isn't a big problem, it just irritated me and it should be easy to fix - and get's you 3% more performance on sending.

ZIP with Air Traces.
[esp32-streamer-csr-client-pdus.zip](https://github.com/espressif/esp-idf/files/1251615/esp32-streamer-csr-client-pdus.zip)

mywang-espressif, 2017-10-09:
Hi @mringwal, good to see you again. The 26-byte fragment issue that you describe is exact the case. And as I look into the code, there is a workaround to solve another issue related to AES half word alignment. Therefore when the data is to be fragmented, the fragment size will be an even number.

mringwal, 2017-10-09:
Good to see you again, too, @mywang-espressif. Will you try to fix it? It would be an easy performance improvement (aka "low hanging fruit" :)

Of course, the 'connection interval only partially used' (https://github.com/espressif/esp-idf/issues/936) is more important, but your colleague @Weijian-Espressif seem to have started looking into this.

mywang-espressif, 2017-10-16:
@mringwal , this bug seems difficult to fix as it is related to hardware issue. I will look into this later to figure it out whether the payload length limit can be resolved. I am afraid it will take time.

mringwal, 2017-10-16:
@mywang-espressif thanks for checking. While this issue is performance related (and looked simple to fix from the outside :), the other issue #936 is much more important. If I could choose, it would be great if you could help with that one as causes throughput to drop by 50% in many cases.