espressif/esp-idf #939: Close client-server connection OTA_example_main.c

Renata27, 2017-08-28:
Hi @igrr and @nkolban,

I get OTA_example_main.c working perfectly, thank you guys!
But as I have SNTP server, Curl functions and server to host my OTA.bin,
I am having issues related to closing servers.
I have the `sntp.stop(); `which closes SNTP.
But, after I download the OTA_example_main.c from my server and the code runs and perform:
`curl_easy_setopt(handle, CURLOPT_URL, url);`
I don't get curl working any more. The  issue is:
`curl_easy_perform() failed: couldn't resolve host name`

So, I was wondering if `close(socket_id);` closes the client-server connection or keeps it alive...
If alive, How could I close the connection?

```
...
ESP_LOGI(TAG, "Connection closed, all packets received");
 close(socket_id);
...
 ESP_LOGI(TAG, "Prepare to restart system!");
 esp_restart();
```

Thank you guys again for helping!

**Using:**
ESP IDF v2.1
Eclipse IDE
ESP32 core board v2

nkolban, 2017-08-28:
@Renata27 Unfortunately, I have no immediate words of wisdom relating to OTA.  That's a whole area I have never tried before.  Can you describe the intent and design of what you are doing?  I am assuming OTA brings in a new "image", loads that into a distinct area of flash and then reboots the device?  Is that also your understanding/thinking?

dsiganos, 2017-08-29:
Looking at the last two statements and if socket_id is a TCP socket:
```
close(socket_id);
esp_restart();
```

I would expect it to not work because the TCP socket close is a 4 way handshake and it probably does not have the time to do a 4 way handshake. It might not even have the time to send the first packet depending on implementation details.

Renata27, 2017-08-29:
Hi @nkolban and @dsiganos , thank you for your support!

Yes @nkolban , it is working! And it is how you said! I am using the code provided on examples from ESP IDF v2.1 which gives for us this diagram:

[image]

I was just wondering which API will than completely close the connection and not only the socket.
Because I am connecting to other servers (SNTP and http server).
So, I get the time and `stop(sntp);` I send data to a database via curl (with your examples) and when it's time, I perform a GET server on my http server in the cloud. The ota.bin is downloaded, flashed, Esp is rebooted and when the code starts again, curl doesn't work anymore. So I presume that's because socket remains open and somehow I can't send a URL using curl anymore.

So @dsiganos , what would you suggest to do before Esp restarts running the new .bin downloaded via OTA?

Thank you guys for helping out!

Renata27, 2017-08-29:
Guys,

I get this error:

```
C:/Users/Renata/Desktop/testcurl/esp-idf-v2.1/components/freertos/event_groups.c:289 (xEventGroupWaitBits)- assert failed!
abort() was called at PC 0x4008660f on core 0
0x4008660f: xEventGroupWaitBits at C:/Users/Renata/Desktop/testcurl/esp-idf-v2.1/components/freertos/event_groups.c:485

Backtrace: 0x40086b38:0x3ffba670 0x40086c37:0x3ffba690 0x4008660f:0x3ffba6b0 0x400e2408:0x3ffba6e0 0x400d26aa:0x3ffba9c0
0x40086b38: invoke_abort at C:/Users/Renata/Desktop/testcurl/esp-idf-v2.1/components/esp32/panic.c:519

0x40086c37: abort at C:/Users/Renata/Desktop/testcurl/esp-idf-v2.1/components/esp32/panic.c:519

0x4008660f: xEventGroupWaitBits at C:/Users/Renata/Desktop/testcurl/esp-idf-v2.1/components/freertos/event_groups.c:485

0x400e2408: app_main at C:/Users/Renata/Desktop/ota/main/ota_example_main.c:667 (discriminator 1)

0x400d26aa: main_task at C:/Users/Renata/Desktop/testcurl/esp-idf-v2.1/components/esp32/cpu_start.c:331

Rebooting...
```

dsiganos, 2017-08-29:
Regards, the close-reboot sequence, in the simplest case and for prototyping purposes, and, to check that the lack of time is your problem, a delay of a few seconds between teh close and the reboot would probably do it.

A better solution but more complicated is to use the SO_LINGER TCP option with a timeout to make sure you don't wait for ever. I don't know if lwip supports SO_LINGER.

Alternatively, if for some reason the closing of the socket is complicated, you could  use the opening of another socket to signal the event you want to signal.

Renata27, 2017-08-31:
Hi @dsiganos  and @nkolban,

the last error was related to WiFi initialization... The problem is that `esp_restart();` is not like a `reset_cpu` performed by the onboard button. So, I needed to reinitialize WiFi after running from the downloaded OTA.bin.

So I ask you guys if there is an API for reset cpu and not only reboot.

Best regards!

Renata27, 2017-09-08:
Hi guys, everything works fine now! Thank you for supporting!
I used the already avaiable WiFi APIs to reconnect when disconnected or when something
goes wrong.. that's kind a must have.

See you!

closed 2017-09-08 by Renata27