espressif/esp-idf #940: UartRxString  block  xTimer

gdyshi, 2017-08-29:
when I start a new task to use UartRxString, the xTimer with auto-reload mode can't work.
void uart_task()
{
    recv_idx = 0;
    while(1) {
        UartRxString((uint8_t*) buf, sizeof(buf) - 1);// 导致timer无法使用
        int len = strlen(buf);
        memcpy(&recv_buf[recv_idx], buf, BUF_SIZE);
        recv_idx += len;        
        len = decode_buf(recv_buf, &recv_idx);        
    }
}

igrr, 2017-08-29:
Hi @gdyshi,
`UartRxString` is a blocking function, which means that it busy-waits until a complete line has been received from UART. If you are starting `uart_task` as a task with any priority higher than 1, `uart_task` will keep running, preventing FreeRTOS timer task (which has priority 1) from doing anything.

You need to do one of the following:

- (better) Replace UartRxString with a function which does not busy-wait. This can be achieved using UART driver. See UART examples for details. Alternatively, take a look at the recently added examples/system/console example, which shows how to receive commands from UART.

- (worse, but quick) Reduce the priority of uart_task to 0. That will allow timer task to run. However you will still burn CPU cycles waiting for the UART input, which is not necessary.

closed 2017-08-29 by igrr

gdyshi, 2017-08-29:
@igrr 
Thanks for your reply
I want to use the log uart port to send some important(security) commands from pc. In this case, our product has only one sample interface——usb.
Now the log uart port uses one uart port, i'm warrieied that the uart port which is used by log is exclusive. I can't use the log uart port in my task.
Is there any solution to use the log uart port for both log and my task?

igrr, 2017-08-29:
Regarding using the same console for log and for command input — indeed there is a bug in the current IDF which blocks UART writes when UART read is in progress. We have the bug fix in review now, expect to see it on Github in the next couple of days.