espressif/esp-idf #941: wifi: bcn_timout,ap_probe_send_start

rinripper, 2017-08-29:
Hi, 

I´m sending UDP datagrams every 60ms with a ESP32 and receiving them with another ESP32. After 3 minutes (aprox) the ESP disconnect of the network and show:

`I (131565) wifi: bcn_timout,ap_probe_send_start
I (134066) wifi: ap_probe_send over, resett wifi status to disassoc
I (134066) wifi: state: run -> init (1)
I (134067) wifi: pm stop, total sleep time: 0/119579415

I (134068) wifi: n:6 0, o:6 2, ap:255 255, sta:6 2, prof:1
I (134076) PRUEBA: SYSTEM_EVENT_STA_DISCONNECTED`

I upload a little example.

The Tx program is tx_udp.c  and the Rx program is rx_udp.c
[tx_rx_udp.zip](https://github.com/espressif/esp-idf/files/1259104/tx_rx_udp.zip)

I hope you can help me, please. 

Thanks in advance.
Mauricio Marcano

shirish47, 2017-10-30:
```
I (44982) wifi: Init rx ampdu len mblock:7
I (44982) wifi: Init lldesc rx ampdu entry mblock:4
I (44982) wifi: wifi power manager task: 0x3ffdb944 prio: 21 stack: 2560
I (44992) SPEC: start the WIFI SSID:[UNICORN] password:[******] 
I (45002) wifi: wifi timer task: 3ffdbba4, prio:22, stack:3584
I (45012) wifi: mode : sta (30:ae:a4:05:c8:10)
I (45022) SPEC: SYSTEM_EVENT_STA_START 
I (45142) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (46122) wifi: state: init -> auth (b0)
I (46122) wifi: state: auth -> assoc (0)
I (46132) wifi: state: assoc -> run (10)
I (46152) wifi: connected with UNICORN, channel 1
I (46892) event: ip: 192.168.1.121, mask: 255.255.255.0, gw: 192.168.1.1
I (46892) SPEC: WIFI got ip.
I (56132) wifi: pm start, type:0

I (56802) wifi: bcn_timout,ap_probe_send_start
I (59302) wifi: ap_probe_send over, resett wifi status to disassoc
I (59302) wifi: state: run -> init (1)
I (59302) wifi: pm stop, total sleep time: 0/3170926

I (59302) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (59312) SPEC: WIFI DISCONNECTED.

```

I get same issue it connects and just after few seconds it disconnects.

ricksondpenha, 2017-11-29:
@rinripper Hi any update on this issue? any workaround?

jack0c, 2018-01-14:
Seems that you have erase the flash include NVS. Maybe you erase the phy calibration data, so ESP32 will do calibration again, so it works again. Maybe your calibration data was corrupted on some version with NVS bug .
@ricksondpenha @rinripper @shirish47 Things you can do to check what's the problem:
1.check the version, give the commit ID here.
2.dump your NVS data to a file, give the file here.
3.erase the flash and flash again.
4. check menuconfig about Components config->PHY->Do phy calibration ..., you can do phy calibration every time.

my-abousamra, 2018-01-31:
I face this problem when I enable bluedroid stack. when I disable it wifi works normally

jack0c, 2018-03-22:
@ricksondpenha @rinripper @shirish47 Does anything help you?

jack0c, 2018-03-22:
@my-abousamra If it is related with bluedroid, I will consider it as a know issue.

my-abousamra, 2018-03-22:
@jack0c I don't know the root cause

shirish47, 2018-03-23:
@jack0c I don't know why I had this issue but I think it was resolved later. I would suggest you use latest IDF.

jack0c, 2018-03-27:
@ricksondpenha @rinripper If no more response，I will close this issue.

closed 2018-04-04 by FayeY