espressif/esp-idf #942: Bluetooth Controller: Host Buffer Size Command with ACL packet size = 512 bytes fails

mringwal, 2017-08-29:
When Host Controller to Host Flow control is enabled, the Bluetooth Controller needs to segment ACL packets. When using a smallerACL packet size value in the Host Buffer Size command, e.g. 512 bytes instead of 1024, the Bluetooth Controller in the ESP32 returns an Invalid Parameter error.

I was optimizing BLE throughput and without DLE and a connection interval of 30 ms, up to 44 packets could be received in one connection interval. To not limit throughput, I wanted to increase the number of Host ACL packets, so that more could be stored before they get processed. Of course, large 3-DH5 Classic ACL packets would need to get segmented as well, but they will be quickly put together by BTstack.

My best guess is that segmentation of ACL packets is not implemented yet and the Host Buffer Size Command returns an error if segmentation might become necessary. Could you please implement this feature after the other issues that impact stability (#924) and performance (#935, #936) are addressed?

From the Bluetooth 5.0 spec - Host Buffer Size Command:
"The Host_Buffer_Size command is used by the Host to notify the Controller about the maximum size of the data portion of HCI ACL and synchronous Data Packets sent from the Controller to the Host. The Controller shall segment the data to be transmitted from the Controller to the Host according to these sizes, so that the HCI Data Packets will contain data with up to these sizes..."

Log:
[00:00:01.386] CMD => 03 0C 00
[00:00:01.393] EVT <= 0E 04 05 03 0C 00
[00:00:01.397] CMD => 01 10 00
[00:00:01.400] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03
[00:00:01.405] LOG -- hci.c.1802: Manufacturer: 0x0060
[00:00:01.413] CMD => 14 0C 00
[00:00:01.416] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.484] LOG -- hci.c.1716: local name:
[00:00:01.488] LOG -- hci.c.1511: Received local name, need baud change 0
[00:00:01.497] CMD => 02 10 00
[00:00:01.500] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.520] LOG -- hci.c.1812: Local supported commands summary 0x3f
[00:00:01.529] CMD => 09 10 00
[00:00:01.532] EVT <= 0E 0A 05 09 10 00 C4 8B 00 C4 0A 24
[00:00:01.537] LOG -- hci.c.1764: Local Address, Status: 0x00: Addr: 24:0A:C4:00:8B:C4
[00:00:01.548] CMD => 05 10 00
[00:00:01.551] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00
[00:00:01.556] LOG -- hci.c.1733: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:01.569] CMD => 03 10 00
[00:00:01.572] EVT <= 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87
[00:00:01.578] LOG -- hci.c.1791: Packet types cc18, eSCO 1
[00:00:01.583] LOG -- hci.c.1794: BR/EDR support 1, LE support 1

Host Buffer Size: 10 x 512 bytes ACL + 10 x 60 bytes SCO packets
[00:00:01.592] CMD => 33 0C 07 00 02 3C 0A 00 0A 00

Command Complete [0C33] - Unsupported Feature or Parameter Value - Host Buffer Size
[00:00:01.596] EVT <= 0E 04 05 33 0C 11

mywang-espressif, 2017-10-16:
Hi mringwal, your guess is correct. The segmentation of ACL packets from controller to host is not implememented. For now Host_Buffer_Size command with ACL packet size <1021 bytes will be rejected. 1021 is the MAX payload size for 3-DH5 packet type. I cannot say for sure whether the feature can help improve the throughput as you required, and I will evaluate it later.

mringwal, 2017-10-16:
hi. I don't see how this could have a negative impact on performance. The Host has to recombine the ACL packets anyway.

I think it would be good to document this and other oddities with the ESP32 Bluetooth Controller in a central place. For one, it might prevent that others have to post issues here, and it also might save people's time, as they run into this. If e.g. the esp-idf doc / VHCI API says: "The Host Buffer Size Command only allows for ACL bytes >= 1021" nobody should try with a lower value.