espressif/esp-idf #943: Unit Testing not possible: unity.h not found

bmetman, 2017-08-29:
I've been trying to unit test the ESP32 microcontroller by following the [guide](http://esp-idf.readthedocs.io/en/latest/api-guides/unit-tests.html), however when running the following command:

`make TESTS_ALL=1`

I get the following result:

```
user@87289f94a4e1:~/esp/ESP32_firmware$ make TESTS_ALL=1
CC build/LED_test/test_LED.o
/home/user/esp/ESP32_firmware/main/LED/test/./test_LED.c:11:25: fatal error: unity.h: No such file or directory
compilation terminated.
/home/user/esp/esp-idf/make/component_wrapper.mk:228: recipe for target 'test_LED.o' failed
make[1]: *** [test_LED.o] Error 1
/home/user/esp/esp-idf/make/project.mk:392: recipe for target 'component-LED_test-build' failed
make: *** [component-LED_test-build] Error 2
```

So it can' t find the unity header. I've tried looking. For reference, the code I made for testing looks like this:

```
#include "../LED.h"
#include "unity.h"

TEST_CASE("test_LED", "LED"
{
	TEST_ASSERT_EQUAL_INT(2, 2);
}
```
I've tried digging around the configuration option using **make menuconfig** but couldn' t find any options. The test file itself is in a subdirectory called _test_ in the specific component, and a **component.mk** file is created there with the line found in the testing guide. 

I don't know what I'm doing wrong here. If you could help me, that would be greatly appreciated.

mh-dev, 2017-08-30:
I wrote a few weeks ago a short guide how to write re-usable and test able components.
You can find it at https://github.com/mh-dev/esp32_reusable_components.

The for your important part should be to set 
`export EXTRA_COMPONENT_DIRS="$IDF_PATH/tools/unit-test-app/components"`

bmetman, 2017-09-01:
Thank you for your help. I've been able to run the following command now:

`make flash TESTS_COMPONENT='LED' monitor'`

However, when the device reboots, my normal application (as defined in **app_main()** function of the firmware) runs without showing the test menu with all available tests. 

Am I executing the wrong command? Do I need to set an environment variable or a configuration to show the test menu? Many thanks.

mh-dev, 2017-09-02:
I develop component based and have an extra project where I develop them and have the unit tests. I am not aware of a way to do this differently. But I think someone in the forum tries to create a custom build target.

igrr, 2017-09-06:
@bmetman Please check [this forum post](https://esp32.com/viewtopic.php?f=13&t=2867&p=13585#p13466) and see if it answers your question.

closed 2017-11-06 by bmetman

Bendikos, 2024-10-27:
[image]