espressif/esp-idf #944: sntp_init() causes crash

gregstewart90, 2017-08-29:
I updated to the latest IDF and calling sntp_init() now causes the esp32 to constantly reset with the following output.

`

Backtrace: 0x400884d9:0x3ffe0180 0x400875c7:0x3ffe01a0 0x40089585:0x3ffe01d0 0x400898fa:0x3ffe03a0 0x40089269:0x3ffe03e0 0x400893cc:0x3ffe0460 0x4008202e:0x3ffe0480 0x400884d6:0x3ffe0540 0x400884d6:0x3ffe0560 0x40089585:0x3ffe0590 0x400898fa:0x3ffe0760 0x40089269:0x3ffe07a0 0x400893cc:0x3ffe0820 0x4008202e:0x3ffe0840 0x400884d6:0x3ffe0900 0x400884d6:0x3ffe0920 0x40089585:0x3ffe0950 0x400898fa:0x3ffe0b20 0x40089269:0x3ffe0b60 0x400893cc:0x3ffe0be0 0x4008202e:0x3ffe0c00 0x400884d6:0x3ffe0cc0 0x400884d6:0x3ffe0ce0 0x40089585:0x3ffe0d10 0x400898fa:0x3ffe0ee0 0x40089269:0x3ffe0f20 0x400893cc:0x3ffe0fa0 0x4008202e:0x3ffe0fc0 0x4008904d:0x3ffe1080 0x4008904d:0x3ffe10a0 0x40085e89:0x3ffe10c0 0x400fe3a2:0x3ffe1100 0x400fe69e:0x3ffe1120 0x400f2294:0x3ffe1140 0x40176cb7:0x3ffe1160 0x40176d83:0x3ffe1180 0x40176dfd:0x3ffe11c0 0x4010544f:0x3ffe11e0 0x4010586c:0x3ffe1220
0x400884d9: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x400875c7: uxTaskGetSnapshotAll at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:5054 (discriminator 2)

0x40089585: esp_core_dump_write at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:112

0x400898fa: esp_core_dump_to_uart at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:534

0x40089269: commonErrorHandler at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x400893cc: xt_unhandled_exception at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x4008202e: _xt_user_exc at xtensa_vectors.o:?

0x400884d6: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x400884d6: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x40089585: esp_core_dump_write at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:112

0x400898fa: esp_core_dump_to_uart at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:534

0x40089269: commonErrorHandler at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x400893cc: xt_unhandled_exception at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x4008202e: _xt_user_exc at xtensa_vectors.o:?

0x400884d6: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x400884d6: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x40089585: esp_core_dump_write at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:112

0x400898fa: esp_core_dump_to_uart at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:534

0x40089269: commonErrorHandler at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x400893cc: xt_unhandled_exception at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x4008202e: _xt_user_exc at xtensa_vectors.o:?

0x400884d6: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x400884d6: prvTaskGetSnapshot at /Path/to/IDF/esp32/esp-idf/components/freertos/./tasks.c:4994

0x40089585: esp_core_dump_write at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:112

0x400898fa: esp_core_dump_to_uart at /Path/to/IDF/esp32/esp-idf/components/esp32/./core_dump.c:534

0x40089269: commonErrorHandler at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x400893cc: xt_unhandled_exception at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x4008202e: _xt_user_exc at xtensa_vectors.o:?

0x4008904d: invoke_abort at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x4008904d: invoke_abort at /Path/to/IDF/esp32/esp-idf/components/esp32/./panic.c:546

0x40085e89: xQueueGenericReceive at /Path/to/IDF/esp32/esp-idf/components/freertos/./queue.c:2034

0x400fe3a2: sys_mutex_lock at /Path/to/IDF/esp32/esp-idf/components/lwip/port/freertos/sys_arch.c:541

0x400fe69e: sys_arch_protect at /Path/to/IDF/esp32/esp-idf/components/lwip/port/freertos/sys_arch.c:541

0x400f2294: pbuf_free at /Path/to/IDF/esp32/esp-idf/components/lwip/core/pbuf.c:1186

0x40176cb7: sntp_send_request at /Path/to/IDF/esp32/esp-idf/components/lwip/apps/sntp/sntp.c:602

0x40176d83: sntp_request at /Path/to/IDF/esp32/esp-idf/components/lwip/apps/sntp/sntp.c:602

0x40176dfd: sntp_init at /Path/to/IDF/esp32/esp-idf/components/lwip/apps/sntp/sntp.c:602

0x4010544f: startSNTP at /Path/to/IDF/esp32/workspace/cp25g1/main/i2c/ds1339/ds1339.c:31

0x4010586c: task_ds1307 at /Path/to/IDF/esp32/workspace/cp25g1/main/i2c/ds1339/ds1339.c:177

`

igrr, 2017-09-07:
Please try latest master; the core dump looping issue has been resolved in 8433fed7f441121cb65f70a893590eec390d5cda.

FayeY, 2017-09-18:
Hi @gregstewart90 , have you tried the latest IDF? Is this issues solved?

gregstewart90, 2017-09-18:
I've tried that latest IDF, and it does work. It doesn't constantly reset, but you must have a connection to the network before you call sntp_init(). In a previous IDF, I imagine sntp_init() waited for a connection. If you call sntp_init() before a connection is made, the esp32 resets.

closed 2017-11-21 by FayeY