espressif/esp-idf #945: Document and Support I2S LCD mode.

OtherCrashOverride, 2017-08-30:
The current ESP32 Technical Reference Manual describes the I2S LCD mode of operation.  However, it is not clear how to make use of it.  There needs to be better documentation or sample code provided that demonstrates its use.  Alternatively, the current I2S driver could be modified to support the mode.

Any tips, tricks, links, etc to getting this working with 8 bit parallel GPIO output and WS strobe would be appreciated.

frnjjq, 2018-06-09:
I agree. It is quite hard to get to work the I2S LCD mode. It is not included in the I2S driver which makes it even harder.

koobest, 2018-10-29:
Hi, @OtherCrashOverride, @frnjjq 
the following link provides the i2s-lcd driver
https://github.com/espressif/esp-iot-solution/tree/master/components/i2s_devices/

closed 2018-10-30 by FayeY