espressif/esp-idf #948: Re-Declaration of list_node

lucazader, 2017-09-01:
I have a protocol handler that will process messages that are sent both over bluetooth and over wifi.

A recent change in de100327 has meant that there is now a re-declaration of `list_node` causing my code not to compile. (as seen below).

```
In file included from C:/0000-Working/ESP32/esp-idf/components/bt/bluedroid/osi/include/fixed_queue.h:23:0,
                 from C:/0000-Working/ESP32/esp-idf/components/bt/bluedroid/hci/include/hci_layer.h:24,
                 from C:/0000-Working/ESP32/esp-idf/components/bt/bluedroid/device/include/controller.h:28,
                 from C:/0000-Working/ESP32/esp32.core/main/gatt_client.h:16,
                 from C:/0000-Working/ESP32/esp32.core/main/protocol.c:41:
C:/0000-Working/ESP32/esp-idf/components/bt/bluedroid/osi/include/list.h:108:7: error: 'list_node' redeclared as different kind of symbol
 void *list_node(const list_node_t *node);
       ^
In file included from C:/0000-Working/ESP32/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:48:0,
                 from C:/0000-Working/ESP32/esp-idf/components/esp32/include/esp_event.h:23,
                 from C:/0000-Working/ESP32/esp-idf/components/esp32/include/esp_wifi.h:69,
                 from C:/0000-Working/ESP32/esp32.core/main/protocol.c:18:
C:/0000-Working/ESP32/esp-idf/components/lwip/include/lwip/apps/dhcpserver.h:182:2: note: previous declaration of 'list_node' was here
 }list_node;
  ^
```

code worked fine before this commit.

lucazader, 2017-09-04:
@igrr @projectgus 

Just wondering if there are any comments on this.
Perhaps re-naming one of these to a more file specific name as not to conflict with the other?

projectgus, 2017-09-05:
Hi @lucazader ,

Thanks for reporting this. It's definitely a bug that two structures have such a generic name. Working on a fix now.

Angus

lucazader, 2017-09-05:
Hi @projectgus 

Thanks for the reply.
I look forward to working with the extra stuff in the 3.0 branch again!

lucazader, 2017-09-13:
Hi @projectgus 

Just checking if this is still in the to-do queue, or if it is done waiting to be merged into the github master?

Currently this is the only thing preventing us from switching over to the v3.0 for development

projectgus, 2017-09-14:
Hi @lucazader ,

Sorry for the delay. The fix for this was merged internally yesterday, it should appear on github in the next 24 hours or so (this issue will be updated when it does.)

Angus

closed 2017-09-18 by igrr