espressif/esp-idf #949: RMT isn't working correctly on master

MartyMacGyver, 2017-09-01:
The RMT device is no longer transmitting as expected in transmit mode.

This isn't working for my code:
https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers

Nor is it working against the code my code was originally based upon:
https://github.com/FozzTexx/ws2812-demo

As was reported to me via https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/issues/9, undoing commit 17747b33586075e46cadfe0c096dabb7f1a80df2 rectifies the problem. It's not yet clear whether there is a bug in that commit or if there is extra code required to work successfully based upon it.

For reference, there was some discussion here but it was inconclusive:
https://www.esp32.com/viewtopic.php?f=13&t=2339&p=11273

MartyMacGyver, 2017-09-01:
This may be resolved now - at least, it seems to be working for me.

Both our demos used the old GPIO constuction:

    ....
    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_RMT_CLK_EN);
    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_RMT_RST);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpioNum], 2);
    gpio_matrix_out((gpio_num_t)gpioNum, RMT_SIG_OUT0_IDX + RMTCHANNEL, 0, 0);
    gpio_set_direction((gpio_num_t)gpioNum, GPIO_MODE_OUTPUT);
    ...RMT initialization code follows...

Using the `rmt_set_pin()` instead seems to resolve the issue:

    ....
    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_RMT_CLK_EN);
    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_RMT_RST);
    rmt_set_pin(static_cast<rmt_channel_t>(RMTCHANNEL),
                RMT_MODE_TX,
                static_cast<gpio_num_t>(gpioNum));
    ...RMT initialization code follows...

igrr, 2017-09-07:
Previously, `gpio_set_direction` did not make sure that GPIO function was actually selected in the GPIO matrix. Now it switches to GPIO signal, which undoes the effect of `gpio_matrix_out(..., RMT_SIG_OUT0_IDX + RMTCHANNEL, 0, 0);`. Either switch the order of the two statements, or use `rmt_set_pin` as a higher-level solution, as you did.

closed 2017-09-07 by igrr

MartyMacGyver, 2017-09-07:
Thanks for the extra info!