espressif/esp-idf #951: Unsuccefull git sync ( local to master )

mikekgr, 2017-09-01:
Dear Friends, I need your help. I have esp-idf installed at my computer ( Windows 10 ) and from time to time I am doing sync with the GitHub repository that is https://github.com/espressif/esp-idf.git
In my computer the installed path ( if this matter ) is: E:\msys32\home\kranidis\esp\esp-idf
When I want to do the sync first I cd to the above path and then I execute git giu
Using the gui I do: Remote -< Fetch from -> origin
then
Merge-> Local Merge ( and ofcourse I select master branch...
Last time ( and this has been happen the last two three times ) after the Merge-> Local Merge I see on screen unstaged Changes ( I never change anything inside the esp-idf directory ) and errors like:
Submodule components/esp32/lib 64b0ff41..bf540bee (rewind):
< update wifi lib
< update wifi lib
< update smartconfig to V2.6.8
< update smartconfig to V2.6.8
This problem also completely blocks any other sync with github with a warning/error windows says: You are in the middle of a change ...
What is so wrong? What to do? All the previous times I delete everything at esp-idf directory and I did: git clone --recursive https://github.com/espressif/esp-idf.git
but this is not a solution...
Please try to help me.
Thanks and Best Regards,
Mike Kranidis

lws-team, 2017-09-06:
To get rid of the "unstaged changes", you can just do `git reset --hard`, it will revert to the situation of the last commit you have.

Using a GUI for git is a crutch, you will never really get control of it that way.

igrr, 2017-09-07:
@mikekgr, did the suggestion from @lws-team help resolve the issue?

mikekgr, 2017-09-07:
@igrr  Dear Ivan, I am not sure because, beffore I got the suggestion, I delete all the esp-idf directory and I got it again back for remote. from now on, I use the syggested way { git pull && git submodule update -- } that @me-no-dev has succested me in  https://gitter.im/espressif/arduino-esp32  chat.
So far, I do the sync without any problem. 
For further protection, I made a note that in case of problems, I can use also the sugeested by @lws-team command:  { git reset --hard }.
Thanks for your follow up.
Best Regards,
Mike Kranidis

closed 2017-09-07 by igrr