espressif/esp-idf #952: ADC Callibration

llewellynnu, 2017-09-02:
Is the Callibration valid only for operation with a gain of 0dB? 
According to the data the vref needs t bebetwwen 1000 and 1200 which is valid only for 0db Operation.
* @param[in]   v_ref       true reference voltage of the ADC in mV (1000 to 1200mV). Nominal
 *                          value for reference voltage is 1100mV.
 * @param[in]   atten       attenuation setting used to select the corresponding lookup table
 * @param[in]   bit_width   bit width of ADC
 * @param[out]  chars       pointer to structure used to store ADC characteristics of module
 */

Can I operate in 11dB with a V_REF at VDDA?

Spritetm, 2017-09-02:
Vref is always going to be 1000-1200mV; the attenuation attenuates the *input* so it will be within VRef range.

llewellynnu, 2017-09-02:
@Spritetm Thanks alot! That clears stuff!

closed 2017-09-05 by FayeY