espressif/esp-idf #954: miniterm error in make monitor

eliabieri, 2017-09-03:
when I want to use make monitor I get the following error.
I use Mac OS 10.12.5 and pyserial is up to date.

> Traceback (most recent call last):
  File "/Users/eliabieri/Documents/ESP-IDF/esp-idf/tools/idf_monitor.py", line 568, in <module>
    main()
  File "/Users/eliabieri/Documents/ESP-IDF/esp-idf/tools/idf_monitor.py", line 502, in main
    monitor.main_loop()
  File "/Users/eliabieri/Documents/ESP-IDF/esp-idf/tools/idf_monitor.py", line 269, in main_loop
    self.handle_serial_input(data)
  File "/Users/eliabieri/Documents/ESP-IDF/esp-idf/tools/idf_monitor.py", line 300, in handle_serial_input
    self.console.write_bytes(b)
  File "/usr/local/lib/python3.6/site-packages/serial/tools/miniterm.py", line 63, in write_bytes
    self.byte_output.write(byte_string)
TypeError: a bytes-like object is required, not 'int'
make: *** [monitor] Error 1

eliabieri, 2017-09-03:
I found the solution. 

> When you are using Python 2.x str is a byte array. For Python 3, you must use bytes like this:

> struct.unpack("<h", bytes([b, a]))[0]

closed 2017-09-03 by eliabieri

projectgus, 2017-09-04:
Thanks for updating, @eliabieri . Even though esptool.py works with both Python 2 & 3,, for now we only support Python 2.x in IDF (and associated tools like idf_monitor).

If you need to configure IDF to run "python2" instead of "python", you can do this in menuconfig under "SDK Tools".

hanfengcan, 2017-12-15:
i use py3,but error like this

neilpanchal, 2018-01-31:
@projectgus What are the plans to upgrade and support Python 3? I hope we make progress because Python 3 is now the default in Ubuntu and frankly, it is the way forward.

lausellduane, 2018-02-08:
@eliabieri I got the same error. How do I implement this "For Python 3, you must use bytes like this:
struct.unpack("<h", bytes([b, a]))[0]" in order to solve the problem?

Herschdorfer, 2019-01-06:
since this is not really solved and also a bit misleading, i have created a PR that should solve this issue.

https://github.com/espressif/esp-idf/pull/2918

Herschdorfer, 2019-01-06:
I made a PR but its pointless since there is already work ongoig.

chnaging the build command from:
`python ${IDF_PATH}/tools/windows/eclipse_make.py -j24`

to 

`python2 ${IDF_PATH}/tools/windows/eclipse_make.py -j24`

solved the issue for me.